cdpuk/givenergy-local #1: Connection issue

j1mco, 2022-04-24:
## Version of the custom_component
V1.0.0

## Describe the bug
Will not connect

## Debug log

```text
NoneType: None
2022-04-24 06:59:36 ERROR (SyncWorker_4) [givenergy_modbus] Did not receive expected response type: ReadInputRegistersResponse != ModbusIOException
2022-04-24 06:59:38 ERROR (SyncWorker_4) [givenergy_modbus] Returned base register (0) does not match that from request (60).
2022-04-24 06:59:39 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up givenergy_local platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/givenergy_local/sensor.py", line 192, in async_setup_entry
    PVEnergyTodaySensor(
  File "/config/custom_components/givenergy_local/sensor.py", line 219, in __init__
    self._attr_unique_id = f"{self.coordinator.data.inverter.inverter_serial_number}_{entity_description.key}"
  File "/usr/local/lib/python3.9/site-packages/givenergy_modbus/model/plant.py", line 32, in inverter
    return Inverter.from_orm(self.inverter_rc)
  File "/usr/local/lib/python3.9/site-packages/pydantic/main.py", line 562, in from_orm
    values, fields_set, validation_error = validate_model(cls, obj)
  File "/usr/local/lib/python3.9/site-packages/pydantic/main.py", line 1001, in validate_model
    value = input_data.get(field.alias, _missing)
  File "/usr/local/lib/python3.9/site-packages/givenergy_modbus/model/register_getter.py", line 69, in get
    return getattr(self._obj, key, default)
  File "/usr/local/lib/python3.9/site-packages/givenergy_modbus/model/register_cache.py", line 26, in __getattr__
    val = self[register]
KeyError: IR:183

```

cdpuk, 2022-04-25:
There's an issue in the underlying library that causes occasional glitches when fetching data (that's the `Did not receive expected response type` you're seeing in the logs. Sometimes this can happen on the very first try, in which case HA will report this error. This is expected to be sorted in the near future, but in the mean time, could you try to connect a few more times and see if you can get past this stage of the process?

j1mco, 2022-04-27:
Tried a few more times.  Logs maybe more legible:

`Error while setting up givenergy_local platform for sensor
20:24:20 – (ERROR) Sensor (custom integration)
Did not receive expected response type: ReadHoldingRegistersResponse != ReadInputRegistersResponse
20:24:19 – (ERROR) /usr/local/lib/python3.9/site-packages/givenergy_modbus/modbus.py - message first occurred at 20:24:16 and shows up 2 times
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 164 bytes (0 received)
20:24:19 – (ERROR) /usr/local/lib/python3.9/site-packages/givenergy_modbus/modbus.py
Transaction failed
20:24:19 – (ERROR) /usr/local/lib/python3.9/site-packages/givenergy_modbus/transaction.py
Returned base register (60) does not match that from request (120).
20:24:17 – (ERROR) /usr/local/lib/python3.9/site-packages/givenergy_modbus/modbus.py`

lxalln, 2022-07-07:
I'm getting the same issue, I cannot connect. I've tried dozens of times.
I also have GivTCP running on the network, and that is working (mostly) fine, so the inverter is online and available.

Logs:
```
Logger: givenergy_modbus
Source: /usr/local/lib/python3.9/site-packages/givenergy_modbus/modbus.py:82
First occurred: 10:40:33 (14 occurrences)
Last logged: 10:45:39

Returned base register (60) does not match that from request (0).
Returned base register (0) does not match that from request (180).
Returned base register (0) does not match that from request (60).
Returned base register (60) does not match that from request (120).
Returned base register (60) does not match that from request (180).
```

```
Logger: givenergy_modbus
Source: /usr/local/lib/python3.9/site-packages/givenergy_modbus/modbus.py:94
First occurred: 10:40:14 (18 occurrences)
Last logged: 10:45:40

Did not receive expected response type: ReadInputRegistersResponse != ModbusIOException
Did not receive expected response type: ReadHoldingRegistersResponse != ModbusIOException
Did not receive expected response type: ReadHoldingRegistersResponse != ReadInputRegistersResponse
Did not receive expected response type: ReadInputRegistersResponse != ReadHoldingRegistersResponse
```

lxalln, 2022-07-07:
I'm running 2022.7, it's still not working:

```
Logger: givenergy_modbus
Source: /usr/local/lib/python3.10/site-packages/givenergy_modbus/transaction.py:100
First occurred: 00:08:20 (4 occurrences)
Last logged: 00:09:13
Transaction failed

Logger: givenergy_modbus
Source: /usr/local/lib/python3.10/site-packages/givenergy_modbus/modbus.py:52
First occurred: 00:08:20 (8 occurrences)
Last logged: 00:09:13

    Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 164 bytes (0 received)
    Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

NoneType: None
```

j1mco, 2022-07-28:
Just to add here, I find a connection will fail if there is something else polling the device (givtcp for example)

rjenx, 2022-12-27:
Sorry my repo is private so you wont see the error. I saw this in my logs yesterday, I have not seen the error since and my HA is showing up to date data  today.  I will monitor if this occurs again and under what circumstances.

cdpuk, 2023-01-01:
Closing for now as I don't think there's anything that could be changed in this project to reduce connectivity issues.

closed 2023-01-01 by cdpuk