cdpuk/givenergy-local #2: Instalation for a complete newbe....

mprowe, 2022-04-25:
I am considering a GivEnery system and looing into the monitoring & management of same.
I have come across your [announcement](https://community.home-assistant.io/t/givenergy-solar-inverter/407844) on the Home Assistant site. I have now got the HAOS running as a (test) VM but I'm stuck trying to install your [Monitoring Tool](https://github.com/cdpuk/givenergy-local).
Can you offer any pointers?

Regards, M.

```
System Health
version: core-2022.4.7
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
user: root
virtualenv: false
python_version: 3.9.9
os_name: Linux
os_version: 5.10.108
arch: x86_64
timezone: Europe/London

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: ok

host_os: Home Assistant OS 7.6
update_channel: stable
supervisor_version: supervisor-2022.04.0
docker_version: 20.10.9
disk_total: 31.3 GB
disk_used: 2.4 GB
healthy: true
supported: true
board: ova
supervisor_api: ok
version_api: ok
installed_addons: Terminal & SSH (9.3.0)

dashboards: 1
resources: 0
mode: auto-gen
```

cdpuk, 2022-04-25:
The integration is currently available via [HACS](https://hacs.xyz/docs/setup/prerequisites). Once you've got that installed, this repo needs to be added as a [custom repository](https://hacs.xyz/docs/faq/custom_repositories). 

Admittedly this is quite a clunky process but I'll try to improve it and the instructurions at some point.

Without a real inverter to talk to, unfortunately you're not going to see very much. Happy to answer any questions about what it can/can't do, but in general, any data you can get in the GivEnergy portal, you could also get via this integration. They have a [demo](https://portal.givenergy.cloud/) available if that's useful.

closed 2022-04-25 by cdpuk

mprowe, 2022-04-26:
Thank you @cdpuk,

Okay, I'm moving forward. I have HACS installed, [your repository](https://github.com/cdpuk/givenergy-local) added and the **GivEnergy Local** integration installed. I think!

[image]

My problem now, is that I can't find the _"... then enter the inverter address during the setup steps"_ bit from your **Installation** [note](https://github.com/cdpuk/givenergy-local#installation). 

Regarding your comments about "a real inverter", do you think that just the WiFi dongle (powered in a USB hub) will provide a meaningful (although limited) digital conversation? I was hoping that I may be able to get a loan from a givEnergy distributor ;)

Regards, M.

cdpuk, 2022-04-26:
That's the hard bit done. From here you should be able to go to **Configuration** > **Devices & Services** > **Add Integration**.

I'm not sure what a Wi-Fi dongle would do on its own. You might be the first person to try it though!

In any case, here's some of the data available in a real system:

[image]

mprowe, 2022-04-26:
Thanks once again @cdpuk,

As you say, this is as far as I can go without something to look at. So, I'll go quiet now...
One last question: You mention _"Modbus support is provided by the givenergy-modbus library."_ in the [Device Support](https://github.com/cdpuk/givenergy-local#device-support) section.
Can you tell me a bit more about this? Is it included in the Integration download? Do I need to get it from givEnergy?

Regards, M.

cdpuk, 2022-04-27:
It just means I didn't do the hard work of actually talking to the inverter! The comms are handled by this library: https://github.com/dewet22/givenergy-modbus

Nothing to worry about from your point of view - it's automatically installed for you.