cdpuk/givenergy-local #11: Looking for a definition for "sensor.grid_export_power"

mprowe, 2022-10-26:
**Is your feature request related to a problem? Please describe.**
No. I don't think it is a problem? I am seeking an understanding of the Sensors that GivEnergy.local is retrieving from the inverter.

**A clear and concise description of what you want to happen.**
Of the 2 Devices and 31 Entities available once my connection is established, some (I think) are related to the OB115m (in my installation but I understand that another common power meter is the EM115m).
Can you confirm and perhaps offer a fuller definition on how  these sensors map to the OB115m. In particular the sensor called "sensor.grid_export_power". As I compare this sensor to my reference power meter (Eastron SDM230m), I find that the OB115 has the same value BUT an opposite sign! So, when my SDM230m is recording an input (value is positive), the OB115m is showing the same value BUT as an output (value is negative -).

**Describe the solution you'd like**
An understanding on what I'm seeing...

**Describe alternatives you've considered**
I have checked that OB115m CT is clamped to the AC Live (and not the Neutral) and that the CT arrow is pointing away from the grid.

**Additional context**
Screenshot while battery is charging. Forced a test charge by turning on  "Timed Charge" in the settings tab on "My Inverter" resulting in ~2.5kW INPUT. But the OB115 is showing it as a OUTPUT?
[image]

SciFi-Bob, 2023-02-14:
Hey!

So the sensor is doing what it says.. Though I agree that it is a bit odd :-)

The value is how much you are exporting, so:- 

1kWh means you are sending the grid 1 Kilowatt per hour
-1kWh means you are pulling 1 Kilowatt per hour from the grid.

I have added a custom sensor to get around that and give me a +ve value for the Import value. 

platform: template
  sensors:
    grid_import_power:
      value_template: "{% if states('sensor.grid_export_power') | int < 0 %}{{states('sensor.grid_export_power') | int * -1 }}{% else %}0{% endif %}"
      icon_template: "mdi:transmission-tower-export"
      unit_of_measurement: "W"
      device_class: "power"

If the creator reads this, I have done the same for the battery charge to give me two sensors, one for charging and one for discharging rather than just the single value that can go +ve or -ve

HTH

cdpuk, 2023-03-26:
I hope the above comment has helped to explain. The integration doesn't seek to transform the values reported from the inverter, so in this case if a value is inverted then template sensors can be used to obtain the desired result.

closed 2023-03-26 by cdpuk