cdpuk/givenergy-local #16: Different battery voltage shown in GE app compared to integration

ianfretwell, 2022-12-15:
I have a Giv-Bat 5.2 (80% DOD) with firmware 3009.

Currently the integration is showing battery voltage as 52.016v (with battery charge as 16% & battery percent as 5%)

by comparison the GE app is _(with a working local connection)_ showing:-

50.91v  and also a battery percent of 5%.

Why the difference ?

cdpuk, 2022-12-16:
Thanks for the report - well spotted.

The reason for the discrepancy is that the voltage currently reported in HA is from the BMS, whereas the voltage you see in the app is as the inverter sees it. The difference in the percentage charge sensors is caused by the same thing - you'll see one entity is provided by the Inverter device and the other provided by the Battery device. 

I think the answer will be to represent both values for the battery voltage, like we currently do for the charge percent. The app is not displaying all available values - much more is available when querying the devices at a low level.

As for the reason behind the differences - I can't answer that. That's down to how GivEnergy design their hardware and write their firmware.

closed 2022-12-16 by cdpuk

ianfretwell, 2022-12-16:
Cheers for the explanation!