cdpuk/givenergy-local #19: Grid Export Scheduling

bnuttall, 2023-01-19:
**Is your feature request related to a problem? Please describe.**
I have a tariff with Octopus Energy that has the following features:

- Heavily discounted electricity between 12:30am and 4:30am
- Good payments for feeding in between 4pm and 7pm 

I want to support the following use case:

- Charge the battery to full each night between 12:30am and 4:30am
- Use the battery to offset grid import to the greatest possible extent during the day
- At around 3:30pm, calculate if I'm likely to have a residual charge in the battery at the end of the day given my historical usage information
- If I'm likely to have residual charge, force a discharge to grid starting at 4pm

The issue is that there doesn't seem to be entity(ies) for setting forced grid discharge. These counters are available in the Givenergy cloud, but not in the integration.

**Describe the solution you'd like**
Entity(ies) for timed force export, and services for setting them 

**Describe alternatives you've considered**
Could do it manually in the cloud, but that won't be nicely integrated into Home Assistant.

cdpuk, 2023-01-22:
Hi, thanks for the feedback. I'm fairly confident the integration already has what you're looking for. 
* There's a service called "GivEnergy Local: Activate timed export mode".
* `binary_sensor.battery_discharge_slot_1` indicates whether you're in that slot, and holds the current configured time set by that service as an attribute.
* `sensor.battery_mode` will show the current mode, e.g. Eco, Timed Discharge, Timed Export.
* The service "GivEnergy Local: Enable timed charging" can configure the charge slot and target SOC.
* `binary_sensor.battery_charge_slot_1` behaves much like the discharge version.
* `switch.battery_ac_charging` can turn on A/C charging.

Do those give you what you need?

cdpuk, 2023-01-25:
Closing for now but happy to discusss further.

closed 2023-01-25 by cdpuk

bnuttall, 2023-02-08:
Thanks for the reply. I've spent some more time understanding what you've
made available and you're correct.

Also, thanks for the great integration!