cdpuk/givenergy-local #20: Is it possible to have (or have I missed) a sensor to read "Import Total"

mprowe, 2023-01-22:
[image]

Regards, Martin

cdpuk, 2023-01-22:
This doesn't currently exist but can easily be added. My only word of warning is that I've seen an engineer visit mess up these total values, so if you want reliable long-term statistics you may be better off accumulating the daily values. The energy dashboard does this well automatically, but doesn't give you sensors for the total values.

mprowe, 2023-01-23:
Thank you for the comments. From my POV, I still think it would be useful, as it improves my confidence in the data set.  I will then be able to relate that value it to a SDM230m power meter that I have had installed.
Are you also able to confirm (or correct) my understanding that this sensor ("Import Total") is the relayed reading from the OB115 (or EM115) fitted as part of the GivEnergy install?
Regards, M.

closed 2023-01-25 by cdpuk