cdpuk/givenergy-local #21: Incorrect data reported

alfwro13, 2023-01-23:
I have seen this on several occasions when my battery level is reported empty and a couple moments later it reports back the correct values. This is clearly visible when the data is displayed in a form of a graph.
Thanks

ianfretwell, 2023-01-26:
Yes, I see the same - I just presumed it was due to failures in communicating with the battery itself though...

alfwro13, 2023-01-26:
That is my guess as well - but it would be nice if this was better handled - i.e. instead of recording 0 as it is the case with the sensor.battery_charge it would be better to ignore that value.

JamesSwift, 2023-02-01:
I've had random extreme values come through for energy consumption, battery charge, and battery percentage. For example today my battery was at 32,000% for a short time...

cdpuk, 2023-02-01:
This is indeed a problem. Unfortunately the attempt to fetch data reports success, yet provides garbage values. Some [effort](https://github.com/cdpuk/givenergy-local/blob/master/custom_components/givenergy_local/coordinator.py#L75) has already gone in to detecting truly impossible values so they can be ignored, but perhaps more can be done here. This is surprisingly difficult as values such as 0% SOC, 0W, while unlikely, are technically valid. The SOC > 100% is something that could be detected, but that alone won't be enough to catch them all.

I'll keep an eye on mine and try to work out some options.

JamesSwift, 2023-02-01:
Thank you. I was so happy to find your integration, thanks for all your hard work.

cdpuk, 2023-02-01:
By some luck I had one of these bad data points a few days ago. There's some slightly tweaked logic in v1.6.1 that will improve things a bit. Happy to revisit if it's still a problem.