cdpuk/givenergy-local #22: Battery charge switch doesn't trigger AC charging

ColindHall, 2023-01-24:
Version 1.5.0

## Describe the bug
Battery charge switch isn't taking any affect on the givenergy battery.

Battery details: 102 Ah | 5.22 kWh | (1 Pack)
Battery Firmware Version: 3009
Software version: D0.535-A0.535
Inverter model: GIV-AC3.0
Inverter Firmware Version: D0.535-A0.535

## Debug log

```text

2023-01-24 20:30:57.169 ERROR (SyncWorker_4) [givenergy_modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
2023-01-24 20:30:57.175 ERROR (MainThread) [custom_components.givenergy_local] Function failed Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
```

cdpuk, 2023-01-25:
Possibly two things going on here:
1. The A/C charging switch alone will not enable charging. The current time must also fall within one of the charge slots. This is just how the inverter behaves, rather than any logic baked in to the integration.
2. The inverter connections are notoriously unreliable. For this reason, the integration will attempt up to 5 times to apply a new configuration value before giving up with that error message. Unfortunately there's not much we can do about that, but certainly for my usage it's extremely rare for it not to work witin the 5 attempts.

Unfortunately there's not really anything that can be changed within the integration to fix the error you're seeing.

closed 2023-01-25 by cdpuk

ColindHall, 2023-01-26:
Thanks for the charge slot tip. 

Seems if I use the giv mobile app, I can set it to “solar and grid” and set charge times from 00:00 till 23:59 and the I can use the battery charge switch in home assistant!

Will keep monitoring incase it messes anything else up, but seems to work well at the moment.

Many thanks 
Colin