cdpuk/givenergy-local #24: Disable Eco mode service

alfwro13, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
There is a service to Activate Eco Mode but as far I can tell there is no service to Disable it 

**Describe the solution you'd like**
Create a service that would allow to disable the Eco Mode

**Describe alternatives you've considered**
Currently the only option seems to be to manually got to the inverter and disable that option

thank you

cdpuk, 2023-01-29:
That's true, there is no way to turn off Eco mode. Eco mode means the battery will only discharge to meet demand, which makes sense. While I see it's possible to toggle a boolean value via the low-level settings in the GE portal, this doesn't map cleanly to a mode as you see in the mobile app. I suspect this is because the act of "turn off eco mode" is quite hard to define, and judging by the mobile app, GE would rather you think of the discharge mode as being one of:
* Eco / Dynamic
* Timed discharge
* Timed export

Which is why you'll see there are services to switch between these modes.

What inverter/battery behaviour are you trying to achieve by turning off eco mode? While it would be possible to expose more of these low-level settings, can you achieve what you need via the high-level options that the app provides?

alfwro13, 2023-01-30:
On couple of occasions I have noticed that when the battery if empty and Eco mode is enabled it won't wait for the Time charge time slot and it will start charging the battery right away on the day tariff.

cdpuk, 2023-02-01:
It sounds like that can be achieved with a timed charge configuration for the 0:30 - 07:30 period. Assuming this slot is fixed, this only needs configuring once, with no automation.

On you comment about the battery charging when empty - the default configuration will stop discharging the battery at 4%. It can then drop slowly to 0% over a few hours, at which point it will recharge back up to 4%. Eco mode won't affect this behaviour. Is that what you're seeing?

cdpuk, 2023-02-11:
Closing due to inactivity. Happy to continue discussion.

closed 2023-02-11 by cdpuk