cdpuk/givenergy-local #26: Updated HA giving an error: No module named 'pymodbus.client.sync'

n1ghtwish, 2023-02-05:
This has been working fine but I needed a new version of HA to support Gmail OAUTH integration.

Have you used this with a newer HA version?

Previous Version of HA: 2022.11.05
New Version of HA: 2023.2.2

Upon updating HA, immediately the integration stopped working. Rolling back my HA version fixes it (I run this in a docker so I've created a "legacy" HA instance for now to carry on with GE integration.

I've tried removing and re-adding the HACS Givenergy integration from scratch, actually I can't delete the Integration because the error for that is along the same lines regarding pymodbus.client.sync.

Error when trying to add a new Integration for GivEnergy:

2023-02-05 22:34:58.071 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration givenergy_local: No module named 'pymodbus.client.sync'

Error when trying to delete the old one:

2023-02-05 22:52:02.564 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/lib/python3.10/site-packages/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/lib/python3.10/site-packages/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/lib/python3.10/site-packages/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/lib/python3.10/site-packages/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
  File "/usr/lib/python3.10/site-packages/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/lib/python3.10/site-packages/homeassistant/components/http/view.py", line 145, in handle
    result = await result
  File "/usr/lib/python3.10/site-packages/homeassistant/components/config/config_entries.py", line 86, in delete
    result = await hass.config_entries.async_remove(entry_id)
  File "/usr/lib/python3.10/site-packages/homeassistant/config_entries.py", line 1043, in async_remove
    await entry.async_remove(self.hass)
  File "/usr/lib/python3.10/site-packages/homeassistant/config_entries.py", line 569, in async_remove
    component = integration.get_component()
  File "/usr/lib/python3.10/site-packages/homeassistant/loader.py", line 730, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/givenergy_local/__init__.py", line 10, in <module>
    from .coordinator import GivEnergyUpdateCoordinator
  File "/config/custom_components/givenergy_local/coordinator.py", line 8, in <module>
    from givenergy_modbus.client import GivEnergyClient
  File "/config/.local/lib/python3.10/site-packages/givenergy_modbus/client.py", line 8, in <module>
    from pymodbus.client.sync import ModbusTcpClient
ModuleNotFoundError: No module named 'pymodbus.client.sync'

cdpuk, 2023-02-11:
I have just updated to 2023.2.3 with no issues. Is this still a problem? Reinstalling the integration could be worth a go, but I can't see why that would be necessary. 

I would be curious to see what version of `pymodbus` is installed. It should be possible to see by running `pip list installed` within your HA installation.

n1ghtwish, 2023-02-11:
Hi :)

that command says:

pymodbus                         3.1.1

The docker container I am using is this one: https://hub.docker.com/r/linuxserver/homeassistant

This appears to run this under the hood: https://github.com/linuxserver/docker-homeassistant

I don't know how much you know about docker stuff in general so don't want to be teaching you to suck eggs so to speak, so I'll just mention that it will compose the image with the DockerFile content from this file: https://github.com/linuxserver/docker-homeassistant/blob/main/Dockerfile

cdpuk, 2023-02-11:
That version of `pymodbus` is too new. Something must have forced it to a newer version, as the `givenergy_modbus` library has an explicit requirement on a version < 3.

Try `pip install pymodbus==2.5.3`

n1ghtwish, 2023-02-11:
Docker locks the versions of anything installed to a snapshot in time, the process for updating HA for example is to update the docker image and apply it to the container.

What I can probably do is use the one above as a template but deploy a custom version of it that specifies a lower version of pymodbus.

I suspect the docker image has just had an "apply all the things" update approach used, which normally might be fine, but if HACS or GE modbus has a dependency on an earlier version of pymodbus, then it wouldn't take that into account.

I'll see if I can deploy this tomorrow and report back to confirm it's working!

cdpuk, 2023-02-12:
FWIW my instance runs in Docker, but using the official image: https://hub.docker.com/r/homeassistant/home-assistant. Similarly the upgrade path on that is to simply bin the old image and launch a newer version aginst the same config directory.

If I have time later I'll try to install a test deployment from scratch so I can understand better how HA+HACS handles dependencies.

cdpuk, 2023-02-12:
It looks like the default installed version of `pymodbus` was bumped from 2.5.3 to 3.1.1 in HA 2023.2, which I assume is present due to the core Modbus integration. That said, as soon as I installed HACS and the GivEnergy integration the version was changed back to 2.5.3.

I'm struggling to find documentation that describes the correct thing to do in this case. It's not directly an issue with the integration, since it's the `givenergy_modbus` project that holds the dependency on `pymodbus==2.5.3`. That project is not actively maintained at present, so getting the version uplifted could be challenging.

Since nobody else has had issues, I'm going to close this for now but may have to revisit the issue in the future. If you can provide steps to reproduce using an official HA installation method I'd be keen to know.

closed 2023-02-12 by cdpuk

n1ghtwish, 2023-02-12:
I forgot to reply earlier but I moved over to the official one you linked and it seemed to solve my problem with this :)

I tended to use linuxserver for a lot of the dockers but in this case, the official one seems a better bet.