cdpuk/givenergy-local #31: Failed to connect to inverter

Philhews, 2023-02-22:
I'm trying to connect to my inverter and I get the following log error

`This error originated from a custom integration.

Logger: custom_components.givenergy_local
Source: custom_components/givenergy_local/config_flow.py:27
Integration: GivEnergy Local (documentation, issues)
First occurred: 08:08:28 (14 occurrences)
Last logged: 09:44:42

Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 48, in async_step_user
    serial_no = await read_inverter_serial(self.hass, user_input)
  File "/config/custom_components/givenergy_local/config_flow.py", line 27, in read_inverter_serial
    serial_no: str = plant.inverter.inverter_serial_number
  File "/usr/local/lib/python3.10/site-packages/givenergy_modbus/model/plant.py", line 32, in inverter
    return Inverter.from_orm(self.inverter_rc)
  File "/usr/local/lib/python3.10/site-packages/pydantic/main.py", line 578, in from_orm
    values, fields_set, validation_error = validate_model(cls, obj)
  File "/usr/local/lib/python3.10/site-packages/pydantic/main.py", line 1103, in validate_model
    values = validator(cls_, values)
  File "/usr/local/lib/python3.10/site-packages/givenergy_modbus/model/inverter.py", line 212, in compute_model
    values['inverter_model'] = Model.from_serial_number(values['inverter_serial_number'])
  File "/usr/local/lib/python3.10/site-packages/givenergy_modbus/model/inverter.py", line 39, in from_serial_number
    raise UnknownModelError(f"Cannot determine model number from serial number {serial_number}")
givenergy_modbus.model.inverter.UnknownModelError: Cannot determine model number from serial number EA********
`
I have * out the end of my serial number but the failure comes down to the modbus code not accounting for the EA serial. It should identify as a Gen2 (because that's what I have).

I did try looking for the code ony server to locally modify but the path it lists doesn't seem to exist.

cdpuk, 2023-03-06:
I notice this has very recently been resolved in the `givenergy_modbus` library, but is not yet included in a release. We'll have to wait for that to happen, then this integration can be updated.

https://github.com/dewet22/givenergy-modbus/commit/01224c77bab68309e06c0dd7e946dfe41b28de22

veehexx, 2023-05-04:
I've hit this issue with their GIV-HY3.6 Gen2 inverter that i had installed yesterday. S/N starts with EA.

> Quick and dirty - edit /usr/local/lib/python3.10/site-packages/givenergy_modbus/model/inverter.py change line 26 'ED': cls.Gen2 to 'EA', restart home assistant.

sounds easy enough fix, but i cannot find the inverter.py file either via HAOS SSH, or a suitable container to even start to look at via portainer.

mchugh19, 2023-08-07:
I have a very similar sounding issue

```
2023-08-07 19:14:52.553 ERROR (MainThread) [custom_components.givenergy_local] Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 48, in async_step_user
    serial_no = await read_inverter_serial(self.hass, user_input)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/config_flow.py", line 27, in read_inverter_serial
    serial_no: str = plant.inverter.inverter_serial_number
                     ^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/givenergy_modbus/model/plant.py", line 32, in inverter
    return Inverter.from_orm(self.inverter_rc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "pydantic/main.py", line 577, in pydantic.main.BaseModel.from_orm
  File "pydantic/main.py", line 1102, in pydantic.main.validate_model
  File "/config/deps/lib/python3.11/site-packages/givenergy_modbus/model/inverter.py", line 212, in compute_model
    values['inverter_model'] = Model.from_serial_number(values['inverter_serial_number'])
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/givenergy_modbus/model/inverter.py", line 39, in from_serial_number
    raise UnknownModelError(f"Cannot determine model number from serial number {serial_number}")
givenergy_modbus.model.inverter.UnknownModelError: Cannot determine model number from serial number FD####G###
```

Install was just completed, so it looks like a new inverter generation. Any ideas?

cdpuk, 2023-08-16:
This is another issue that would be fixed by #61

closed 2024-01-29 by cdpuk