cdpuk/givenergy-local #41: Change refresh rate

alfwro13, 2023-03-25:
Is there a way of changing the default refresh rate?

cdpuk, 2023-03-26:
Hi, there is no way to change the refresh rate at present. I'm open to the idea of someone adding this but care must be taken when speeding up the refresh rate as it's not an instantaneous operation and also the time taken to refresh depends on the number of batteries. 

What sort of refresh rate are you looking for? That would be good to know for the benefit of any potential future implementation.

closed 2023-03-26 by cdpuk

alfwro13, 2023-03-27:
Well the main objective better data resolution.
For now I manually change it to 5 seconds in the coordinator.py:

```
class GivEnergyUpdateCoordinator(DataUpdateCoordinator[Plant]):
    """Update coordinator that enables efficient batched updates to all entities associated with an inverter."""

    require_full_refresh = True
    last_full_refresh = datetime.min

    def __init__(self, hass: HomeAssistant, host: str, num_batteries: int) -> None:
        """Initialize my coordinator."""
        super().__init__(
            hass,
            _LOGGER,
            name="Inverter",
            update_interval=timedelta(**seconds=5),**
```
but it would be nice if there was a way of changing that either from the Devices and Services Home Assistant page or via a HA service