cdpuk/givenergy-local #42: State of charge total and percentage

gavanfantom, 2023-03-31:
Currently the individual battery state of charge is available in kWh, but not a combined system total, nor is it available as a percentage. The GivEnergy app shows the battery state for the whole system as a percentage as well as making percentage charge available for individual batteries, and a similar pair of sensors (combined kWh and combined percentage) as well as a percentage per battery would be pretty neat.

cdpuk, 2023-04-13:
I only have one battery, making this difficult to work out. Have you see the differnet SOC sensors?
* `sensor.battery_charge` is reported from the battery. I'd expect this to be the individual pack capacity.
* `sensor.battery_percent` is reported from the inverter. I'd expect this to be calculated across all batteries.

If that's true, great, otherwise more research required to understand GivEnergy's logic.

For the kWh total, could you define a template sensor that sums the kWh values across your batteries? Does that match what the GE app tells you?

gavanfantom, 2023-04-14:
Yes, sensor.battery_charge and sensor.battery_charge_2 are the individual charge states of the batteries. Summing those is what I had in mind.

However, you also mentioned sensor.battery_percent, which I had entirely missed. It turns out that the default UI suppresses battery status entities and as such, I'd never been aware that this integration was already providing exactly what I wanted. So, sorry for the noise!

closed 2023-04-27 by cdpuk