cdpuk/givenergy-local #43: Export and Discharge modes are ineffective. without self.set_shallow_charge()  being configurable.

reidjr, 2023-04-01:
Export and Discharge modes are ineffective. 

in givenergy-modbus

set_mode_storage defaults to 
`self.set_shallow_charge(100)  # r110=100`

and
`self.set_shallow_charge() `

is not exposed through givenergy-local.

I think this means that you never discharge in Timed export /discharge as the battery min SoC is 100%. That is the behaviour I see on my inverter. However I am puzzled nobody has noticed this .... Am I missing something ?

Interestingly - Timed export with set_shallow_charge(*) has behaviour of only trickle charging from PV (294w max on my system). So is max PV export, and dumps remaining battery during the timed period.

Timed discharge set_shallow_charge(100) also trickle charges, BUT set_shallow_charge(<100) allows normal PV charging, just like eco outside the discharge window, which is potentially just the right setting for set and forget forced discharge during lucrative set times like octopus flux, and eco at all other times.

cdpuk, 2023-04-13:
As per the comment on #45 - are you using an EPS? The app documentation suggests `battery_soc_reserve` is only effective when an EPS is connected.

reidjr, 2023-04-14:
No EPS. I will comment more fully on #45

cdpuk, 2023-05-15:
The EPS thing still puzzles me. You're probably right about some sort of flag to indicate the presence of an EPS, but it's not obvious. All the EPS stats are populated for my inverter when looking at the raw register values. I'll close this as I think we've sorted everything else now but for future reference here's what the GE app user guide current says about this
[image]

closed 2023-05-15 by cdpuk

reidjr, 2023-05-15:
> ineffective. without self.set_shallow_charge() being configurable.
> (Issue #43)

It's only a guess, but I think it's a bad explanation in the app. The 
battery reserve is set, and determines where the battery stops 
discharging, either when in eco mode, on demand, or under forced 
export/discharge. My guess is that under mains failure, the reserved 
value is overriden, and the battery  will meet demand down to 4%.  It's 
the only thing that makes sense, as there is no point in a reserve that 
is not allocated when needed for the eps.

Sent via BT Email App

  image 
<https://user-images.githubusercontent.com/24318424/238467970-38484297-1af9-4ab5-be20-ec4d89624726.png>