cdpuk/givenergy-local #47: Export from the battery

alfwro13, 2023-04-25:
With the current state of the integration is it possible to schedule an export from the battery to the grid?
I tried using the "GivEnergy Local: Activate timed export mode" service - I selected my inverter and set the time but that seems to be doing nothing for me.
Thanks

cdpuk, 2023-04-25:
That certainly should work. Here is a working example:
[image]
What inverter and battery model do you have? And can you get the desired behaviour via the GE app or portal?

alfwro13, 2023-04-26:
I tried exactly that and it does set the settings but my inverter does nothing until I go and manually switch off the Eco mode in the GE portal

cdpuk, 2023-04-26:
If the "Activate timed export mode" service executes successfully, it will turn off eco mode. Can we check the sequence of events here:
* Call the service using the HA Developer Tools.
* After 5-10 seconds the "Call Service" button should display a green tick indicating a successful call.
* Go to the GE portal and perform a read operation against the "Enable Eco Mode" property. It should be disabled.

[image]

alfwro13, 2023-04-27:
Thanks, that was very helpful. I found the problem on my end. For the tests, I was using short time frames, starting at 9:00 and finishing at 9:01. However, my inverter's time had drifted, so there was nearly a minute difference between HA and the inverter, which is weird because both HA and the inverter sync time with pool.ntp.org. I updated the inverter's time and allowed additional time for the commands to execute, and that seemed to have done the trick. Now all I have to do is wait for GE to fix the firmware so I can switch to Octopus Flux. 
Thanks again!

closed 2023-04-27 by alfwro13