cdpuk/givenergy-local #58: Setting discharge limit to < 370-382W results in discharge of 370-382W

andynash, 2023-06-30:
## Version of the custom_component

1.8.0

## Configuration

```
service: number.set_value
data:
  value: "0"
target:
  entity_id: number.battery_discharge_power_limit
```

Add your logs here.

Added below, seems there are two log sections in this template?

## Describe the bug

When setting the discharge limit number entity to 0 using the relevant service, the number entity **is** correctly updated to 0 (ditto when setting to any other value). However, for any value less than 370-382 W, the battery continues to discharge (at a rate of between 370 and 382 W), as seen by sensor.battery_power and the GivEnergy mobile app.

Higher discharge values appear to work as expected.

N.b. This occurs regardless of actual consumption. If consumption is higher than 370-382 W, the battery continues discharging at that rate, and grid power increases to fill the gap. I've not been able to observe consumption at less than this, as our baseload is usually higher.

This has been observed while there has been no solar generation. When there is solar generation that reduces or eliminates the unwanted discharge, but obviously that is just obscuring the problem.

It is as though any value for discharge power limit that is less than 370-382 W = 370-382 W.

Logs don't appear to show anything particularly informative I'm afraid. Happy to dig further if you can guide me - I've set givenergy_local to debug and filtered for givenergy using grep, while changing from a value of 3600 to 0 and back to 3600.

## Debug log

```
# tail -f /home/homeassistant/.homeassistant/home-assistant.log|grep givenergy
2023-06-30 04:55:45.123 INFO (SyncWorker_23) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-06-30 04:55:45.124 DEBUG (SyncWorker_23) [custom_components.givenergy_local.coordinator] Performing partial refresh
2023-06-30 04:55:47.375 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 2.252 seconds (success: True)
2023-06-30 04:55:47.384 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:55:47.412 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2023-07-01 00:01:00
2023-06-30 04:55:47.432 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:55:47.452 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_2 at 2023-07-01 00:00:00
2023-06-30 04:56:14.814 DEBUG (MainThread) [custom_components.givenergy_local] Attempting function call (5 attempts left)
2023-06-30 04:56:15.008 INFO (SyncWorker_37) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-06-30 04:56:15.009 DEBUG (SyncWorker_37) [custom_components.givenergy_local.coordinator] Performing full refresh
2023-06-30 04:56:19.512 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 4.504 seconds (success: True)
2023-06-30 04:56:19.523 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:56:19.545 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2023-07-01 00:01:00
2023-06-30 04:56:19.566 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:56:19.585 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_2 at 2023-07-01 00:00:00
2023-06-30 04:56:49.125 INFO (SyncWorker_33) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-06-30 04:56:49.126 DEBUG (SyncWorker_33) [custom_components.givenergy_local.coordinator] Performing partial refresh
2023-06-30 04:56:51.371 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 2.246 seconds (success: True)
2023-06-30 04:56:51.379 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:56:51.399 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2023-07-01 00:01:00
2023-06-30 04:56:51.418 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:56:51.438 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_2 at 2023-07-01 00:00:00
2023-06-30 04:57:16.615 DEBUG (MainThread) [custom_components.givenergy_local] Attempting function call (5 attempts left)
2023-06-30 04:57:16.794 INFO (SyncWorker_12) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-06-30 04:57:16.795 DEBUG (SyncWorker_12) [custom_components.givenergy_local.coordinator] Performing full refresh
2023-06-30 04:57:21.273 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 4.479 seconds (success: True)
2023-06-30 04:57:21.281 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:57:21.301 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2023-07-01 00:01:00
2023-06-30 04:57:21.321 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:57:21.340 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_2 at 2023-07-01 00:00:00
2023-06-30 04:57:51.114 INFO (SyncWorker_43) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-06-30 04:57:51.114 DEBUG (SyncWorker_43) [custom_components.givenergy_local.coordinator] Performing partial refresh
2023-06-30 04:57:53.388 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 2.275 seconds (success: True)
2023-06-30 04:57:53.398 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:57:53.425 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2023-07-01 00:01:00
2023-06-30 04:57:53.446 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:57:53.467 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_2 at 2023-07-01 00:00:00
2023-06-30 04:58:23.137 INFO (SyncWorker_33) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-06-30 04:58:23.138 DEBUG (SyncWorker_33) [custom_components.givenergy_local.coordinator] Performing partial refresh
2023-06-30 04:58:25.413 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 2.276 seconds (success: True)
2023-06-30 04:58:25.435 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:58:25.478 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2023-07-01 00:01:00
2023-06-30 04:58:25.501 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2023-06-30 23:59:00
2023-06-30 04:58:25.521 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_2 at 2023-07-01 00:00:00
```

```

cdpuk, 2023-07-04:
What happens if you use the GivEnergy app to set the max discharge power to a problematic value? 

If that works, it's useful to pull the raw values from the GE web portal to try and spot the differences.

What model of inverter and battery are being used?

reidjr, 2023-07-09:
This is a Gen2 inverter issue, present in firmware up to 909 (latest). For me the minimum value reported is 273,. You can't set 0. Same on givtcp, the phone app, the web portal.

andynash, 2023-07-10:
Yes, I tested this morning and it is a GivEnergy issue, possibly Gen2 only, I couldn't say.

I hadn't seen an issue on charging (I don't have a use for limiting charge speed so don't use that), but on discharge it is present when set through the app too - sorry I should have checked that before posting here.

I've upgraded today to 909 to see if that fixes it, and haven't had a chance to test it again yet, but will do so...

reidjr, 2023-07-10:
Sorry to say, 909 doesn't fix it. However there are new commands now being added to givtcp (sorry) and to the givenergy web portal to apply selective battery pause. Works on 909 at least, and can set either off/charge paused/discharge paused/charge+discharge paused. This command does not change the max charge register, its another control.

andynash, 2023-07-10:
Weird, the amount it decides to discharge at (for my one and only test) is now exactly 300W, so that has changed slightly. But no, I concur, 909 doesn't fix.

I'm sure those new controls will be added to this integration in due course, as they do sound simpler/more direct than modifying the power limits (if you are just wanting to pause completely). Maybe as a select?

@reidjr Do you know if there's a bug/community issue somewhere already for following the power limit issue?

@cdpuk I guess you can close this, or mark 'pending upstream' or whatever suits? Thanks!

reidjr, 2023-07-10:
Not as such. Givenergy know the Gen2 does that, its been discussed in the community forums, but never seen a definitive " it will be fixed" comment. As you have probably founfd on the community forum, theres no FAQ, no stickies and no real way to find out whats going on without reading 3000 post threads..... I digress :-)

andynash, 2023-07-10:
I have indeed noticed that 😂 thanks for the update though 👍

I don't pay much attention to the forums, but will try and poke my nose in more often...

andynash, 2023-10-04:
> Sorry to say, 909 doesn't fix it. However there are new commands now being added to givtcp (sorry) and to the givenergy web portal to apply selective battery pause. Works on 909 at least, and can [...]

@cdpuk Is it going to be possible to add this battery pause feature to the integration? (I'm on 909 and can't see it there yet unless Im missing something)

With Intelligent Octopus now supporting Outgoing Fixed/Agile, there's a good reason to want to prevent solar charging during the day (so that solar is exported instead), so I'm sure it would be useful to many.

Thanks for this excellent and integration by the way, I could try GivTCP which has this feature, but yours seems a simpler and more robust solution so I'd prefer to stick with it if at all possible.

cdpuk, 2023-10-04:
GivTCP has gone the route of forking the `givenergy_modbus` library internally. This means new things can be added, but it's engineered in a way that makes it difficult to cleanly import in to projects like this.

Unless we can get updates pushed back to the main `givenergy_modbus` codebase, the quickest approach is probably to do the same as GivTCP, but then we've got 3 completely separate codebases trying to achieve the same thing.

Also, I'm the owner of a Gen1 inverter, so I have no way to test any of these changes (or benefit from them!). In this particular case I'd argue the workaround of changing the charge/discharge rates is pretty good.

I'm going to close this issue, but feel free to raise specific ideas like the pause feature this as something new - it might just take a while to get around to.

closed 2023-10-04 by cdpuk