cdpuk/givenergy-local #64: Add support for timed pause

andynash, 2023-10-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Currently the only way to prevent the battery from charging (e.g. because you have sufficient SoC and wish to maximise export) is to set the charge rate limit to 0W.

The same applies if you wish to prevent the battery discharging.

However the GivEnergy inverter will not actually limit charge/discharge to 0W, instead a seemingly arbitrary and variable limit will apply (see this [closed issue)](https://github.com/cdpuk/givenergy-local/issues/58).

It appears this is "wontfix" for GivEnergy as they have added a time pause feature which covers this requirement.

givenergy-local does not yet support the timed pause functionality, so setting the rate limit to 0W and accepting the limit will really be c. 300W is necessary.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Implementation of the timed pause feature via a service, in a similar way to the existing timed charge, discharge, export services.

I understand that this may not happen quickly as @cdpuk has a gen 1 inverter (which may not support this feature ?), but I'm happy to help test when the time comes 👍

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

See issue referenced above.

**Additional context**
Add any other context or screenshots about the feature request here.

To put a rough value on this, 300W x 8 hours of sunshine at 15p/kWh export over a year is £131.40.

Assuming you need to import less overnight as a result, at 7.5p, the cost can be offset by half, so £65.70 per annum, plus an unknown community cost of not sharing the energy during daylight hours including evening peak.

[assumes tariff is Intelligent Octopus Go with Outgoing Fixed]

cdpuk, 2024-01-29:
Hi @andynash. With v2.0.0 now released, it should be relatively easy to add this in. Is your offer of testing still open? If so, I can have a go at adding this and create a beta release for review.

On the 4 modes listed, are these available via the GE app or web portal? If so, some screenshots would be handy to understand how best to match the functionality.

andynash, 2024-01-31:
That's great news - yes I'd be happy to test this, thanks!

These settings do not appear to be available via the app, just givenergy.cloud, accessed via the cog icon "Remote Control" in the My Inverter section. Hopefully the following will help:

[image]

[image]

[image]

cdpuk, 2024-01-31:
Ideal, thanks.

There are currently some stability issues reported with 2.x, but this looks relatively easy to add in. It would make sense for this to become a select entity presented in the same way as the GivEnergy portal. Could you send one of each of those commads and confirm what appears in the Remote Control History section of that page? If you check "Show Raw Values" I'm assuming they map on to values 0, 1, 2 and 3 (in that order matching the screenshot).

cdpuk, 2024-02-04:
I've made a bit of a guess at the implementation, which is available as HACS beta release v2.1.0-beta1. 

With no way to test this properly, please give this a try and feed back any issues here. Debug logs while the integration is loaded or reloaded will be useful.

cdpuk, 2024-02-04:
Make that v2.1.0-beta2. Also note that I've only added a selection for the mode at this stage. If that works, I'll add in the time slot configuration.

andynash, 2024-02-05:
Thanks, will try that 👍

However I've updated to v2.1.0-beta2, and despite restarting I don't see any new services or controls - it seems I should see a select?

In terms of your earlier question, I first changed the selector at givenergy.cloud through each pause option in order, and then back to Not Paused. This confirmed Not Paused = 0, Pause Charge = 1, Pause Discharge = 2, Pause Charge & Discharge =3. This had no effect on the operation of the battery (because start and end time were default, i.e. 00:00).

I then set start to 13:00 and end to 14:00 and set the selector to Pause Discharge to confirm it operated as expected which it did, and finally set it to Not Paused again.

[image]

cdpuk, 2024-02-05:
Sounds good. Could I have some debug logs while the integration loads? Part of the change is to get it to recognise the additional capabilities of your inverter, which seems not to work yet.

andynash, 2024-04-05:
Sorry for the delay, I've been super busy! I've just taken a look and seen this error which seems relevant?

```
2024-04-05 13:35:03.242 ERROR (MainThread) [homeassistant.components.select] Error while setting up givenergy_local platform for select
  File "/home/homeassistant/.homeassistant/custom_components/givenergy_local/select.py", line 45, in async_setup_entry
```

Following is the full debug during a reload of the integration:

```
2024-04-05 13:39:35.741 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-05 13:39:35.742 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60)
2024-04-05 13:39:35.742 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60)
2024-04-05 13:39:35.790 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2024-04-05 23:59:00
2024-04-05 13:39:35.832 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2024-04-06 00:01:00
2024-04-05 13:39:35.851 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for discharge_slot_1 at 2024-04-05 23:59:00
2024-04-05 13:39:39.174 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Shutting down
2024-04-05 13:39:39.175 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-04-05 13:39:39.275 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-04-05 13:39:39.494 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 10.0.0.27:8899
2024-04-05 13:39:39.494 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Detectig plant
2024-04-05 13:39:39.511 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x32 base_register=0)
2024-04-05 13:39:39.750 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-05 13:39:39.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)
2024-04-05 13:39:39.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)
2024-04-05 13:39:39.761 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x32 base_register=180)
2024-04-05 13:39:39.990 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-05 13:39:39.990 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=180)
2024-04-05 13:39:39.990 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=180)
2024-04-05 13:39:40.012 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:3/ReadHoldingRegistersRequest(slave_address=0x32 base_register=0)
2024-04-05 13:39:40.259 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-05 13:39:40.260 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=0)
2024-04-05 13:39:40.260 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=0)
2024-04-05 13:39:40.266 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:3/ReadHoldingRegistersRequest(slave_address=0x32 base_register=60)
2024-04-05 13:39:40.500 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-05 13:39:40.501 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=60)
2024-04-05 13:39:40.502 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=60)
2024-04-05 13:39:40.517 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:3/ReadHoldingRegistersRequest(slave_address=0x32 base_register=120)
2024-04-05 13:39:40.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
[... cut ...]
2024-04-05 13:39:47.997 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:6/WriteHoldingRegisterResponse(112 -> 50/0x0032)
2024-04-05 13:39:47.997 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] 2:6/WriteHoldingRegisterResponse(112 -> 50/0x0032)
2024-04-05 13:39:47.998 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:6/WriteHoldingRegisterResponse(112 -> 50/0x0032)
```

andynash, 2024-04-23:
Hi @cdpuk, any thoughts on the debug log? Thanks!

pskillen, 2024-05-02:
I'm also excited for this feature, for the exact same reason as Andy. 

I've spent the day methodically testing every permutation of setting to see if I could achieve it but no, the firmware doesn't allow a mode which prioritises export while minimising import and no battery charge.

I have a Gen 3 5kW inverter and would be willing to test if it'll be helpful

andynash, 2024-05-06:
Hi @cdp, sorry to bug you, as I appreciate you giving up your own time to work on this, and that I myself took a while to get back to you with the logs 😁

Just wondering if you expect to be able to look at this soonish, as if not I need to do some work on automations, to work around  the battery charging when I don't want it to?

andynash, 2024-06-18:
@cdpuk I've just upgraded HA to 2024.6.3, and encountered the other now closed issue that prevented startup of the integration, so have now switched from the beta2 to the latest main release.

I'd be happy to switch back to a beta if you are able to look at this again however, as it would be wonderful to get this capability working 👍

cdpuk, 2024-07-09:
I've finally had a bit of time to take another look based on the feedback here. Thanks for the logs - that's exactly what was needed. The new `v2.1.0-beta3` has been rebased on top of the latest mainline changes so will work across all recent HA versions. More testing & feedback would be very useful.

andynash, 2024-07-10:
Thanks that's great news!

I've installed and now see the Battery Pause Mode select entity. I can change it, but in my testing that didn't appear to change the value on the inverter.

I don't yet see an entity for Pause Battery Start and End Times, should I?

cdpuk, 2024-07-10:
Do you have some logs for the attempted cange of value?

The start/end times won't be there yet, but can be soon.

cdpuk, 2024-07-13:
First attempt at time slot configuration can be found in `v2.1.0-beta4`. Unlike other time slots, these are implemented as `time` entities which allow values to be changed without having to call a service.

SciFi-Bob, 2024-08-15:
While you are doing this time based stuff can you also implement the ability to set the timed charge windows?  

The use case here is when the energy provider has 'cheep times' which ours does from time to time, I want to be able to trigger the battery to charge outside of the usual times.

I spoke to GivEnergy support and they told me that the mobile app does this buy setting the timed charge values to charge now and then puts them back to your set values later.

I want to replicate this functionality in HA.

Thanks

cdpuk, 2024-08-20:
Hi @SciFi-Bob, I'm fairly sure what you're asking for already exists as the "Enable timed charging" service. I currently use this with Octopus Intelligent Go to deal with cheap periods outside of the usual 2330-0530 slot. Perhaps I'm missing something - if so please raise a new topic to keep that discussion separate. I'm keen to focus on Timed Pause here, which is still awiting further testing from someone who can actually use it.

SciFi-Bob, 2024-08-21:
@cdpuk - apologies, I missed that !  would love to see how you have that setup end-2-end as it's also my target

closed 2024-11-09 by cdpuk

cdpuk, 2024-11-09:
This has been merged in with the v2.1.0 release. Testing has been limited, but exposure to a wider audience should help iron out any issues.