cdpuk/givenergy-local #69: Unable to connect to inverter

JRascagneres, 2023-11-22:
Likely a new install issue but would like to open an issue so it can be tracked.

## Version of the custom_component
v1.9.0

## Configuration

[image]
```
Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 48, in async_step_user
    serial_no = await read_inverter_serial(self.hass, user_input)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/config_flow.py", line 27, in read_inverter_serial
    serial_no: str = plant.inverter.inverter_serial_number
                     ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/givenergy_modbus/model/plant.py", line 32, in inverter
    return Inverter.from_orm(self.inverter_rc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pydantic/main.py", line 577, in from_orm
    values, fields_set, validation_error = validate_model(cls, obj)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pydantic/main.py", line 1055, in validate_model
    value = input_data.get(field.alias, _missing)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/givenergy_modbus/model/register_getter.py", line 15, in get
    self.get('inverter_serial_number_1_2'),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/givenergy_modbus/model/register_getter.py", line 69, in get
    return getattr(self._obj, key, default)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/givenergy_modbus/model/register_cache.py", line 26, in __getattr__
    val = self[register]
          ~~~~^^^^^^^^^^
KeyError: HR:013
```

```
Did not receive expected response type: ReadInputRegistersResponse != ReadHoldingRegistersResponse
Did not receive expected response type: ReadHoldingRegistersResponse != ReadInputRegistersResponse
Did not receive expected response type: ReadHoldingRegistersResponse != ModbusIOException
Did not receive expected response type: ReadInputRegistersResponse != ModbusIOException
```

```
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 164 bytes (0 received)
Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
```

## Describe the bug
Doesn't connect and allow integration to start

cdpuk, 2024-01-07:
Please could you confirm your inverter model & generation, battery model, and firmware versions for both?

This is probably an incompatibility with newer firmware, which is being worked on under #61.

cdpuk, 2024-01-28:
Closing due to lack of detail, though likely to be resolved by #74

closed 2024-01-28 by cdpuk