cdpuk/givenergy-local #70: Upgrading to Battery fw 3015+ (and Inverter fw 912+) introduces breaking API change

andynash, 2023-12-07:
## Version of the custom_component

1.9.0

## Configuration

```yaml

No change to existing working GUI configuration.

```

## Describe the bug
It appears the latest GivEnergy battery firmware 3015 also updates the inverter firmware to 912 and modifies the way the modbus API works. The version of givenergy-modbus used by givenergy-local appears to be incompatible with these changes, so upgrading the firmware is a breaking change for givenergy-local.

Apparently this is addressed in a newer pre-release version of givenergy-modbus, [as confirmed in comments](https://github.com/cdpuk/givenergy-local/issues/61#issuecomment-1655664314) to an [existing issue](https://github.com/cdpuk/givenergy-local/issues/61) raised regarding the new givenergy-modbus.

This newer version is already used by givtcp, so switching to that integration is a viable alternative, however givtcp takes a different approach to givenergy-local (including reliance on mqtt as well as providing different entities and services), so migrating is likely to be non-trivial (and users may have other reasons for preferring the approach taken by givenergy-local).

If you do upgrade to 3015 (and 912), the only way to get things working again will be to revert the firmware or to migrate to givtcp.

While this has already been acknowledged as mentioned above, this is more pressing now that 3015 has been released. In addition, others may be unaware of the risk of upgrading to 3015 as I was, so I'm posting this as a separate issue to highlight this and prevent others making the same mistake.

## Debug log

```text

2023-12-07 12:22:10.232 INFO (SyncWorker_36) [custom_components.givenergy_local.coordinator] Fetching data from 10.0.0.27
2023-12-07 12:22:10.232 DEBUG (SyncWorker_36) [custom_components.givenergy_local.coordinator] Performing full refresh
2023-12-07 12:22:12.239 ERROR (SyncWorker_36) [givenergy_modbus] Transaction failed
2023-12-07 12:22:12.239 ERROR (SyncWorker_36) [givenergy_modbus] Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 164 bytes (0 received)
NoneType: None
2023-12-07 12:22:12.240 ERROR (SyncWorker_36) [givenergy_modbus] Did not receive expected response type: ReadInputRegistersResponse != ModbusIOException
2023-12-07 12:22:14.749 ERROR (SyncWorker_36) [givenergy_modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
2023-12-07 12:22:14.749 ERROR (SyncWorker_36) [givenergy_modbus] Did not receive expected response type: ReadInputRegistersResponse != ModbusIOException
2023-12-07 12:22:17.256 ERROR (SyncWorker_36) [givenergy_modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
2023-12-07 12:22:17.257 ERROR (SyncWorker_36) [givenergy_modbus] Did not receive expected response type: ReadHoldingRegistersResponse != ModbusIOException
2023-12-07 12:22:19.763 ERROR (SyncWorker_36) [givenergy_modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
2023-12-07 12:22:19.764 ERROR (SyncWorker_36) [givenergy_modbus] Did not receive expected response type: ReadHoldingRegistersResponse != ModbusIOException
2023-12-07 12:22:20.231 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Error communicating with API: 
2023-12-07 12:22:20.232 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 10.001 seconds (success: False)
2023-12-07 12:22:22.269 ERROR (SyncWorker_36) [givenergy_modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
2023-12-07 12:22:22.269 ERROR (SyncWorker_36) [givenergy_modbus] Did not receive expected response type: ReadHoldingRegistersResponse != ModbusIOException
2023-12-07 12:22:24.776 ERROR (SyncWorker_36) [givenergy_modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
NoneType: None
2023-12-07 12:22:24.777 ERROR (SyncWorker_36) [givenergy_modbus] Did not receive expected response type: ReadInputRegistersResponse != ModbusIOException
2023-12-07 12:22:25.292 ERROR (MainThread) [custom_components.ics.sensor]

```

cdpuk, 2023-12-07:
Thanks for the heads up. Can you confirm your inverter and battery models?

For some time this seems to have only affected newer kit but I've noticed my Gen1 5.0 + 8.2 battery is reporting a firmware update available. Currently on inverter D0.450-A0.451 and battery 3012.

andynash, 2023-12-08:
Yes I have a Gen 2 5kW Hybrid inverter and a Gen 2 9m5kWh battery.

Like you I was on 3012 and all was well. When I upgraded to 3015, I didn't notice any warning that the inverter would also be upgraded from 909 (I think) to 912 as part of the update. Its possible I missed something though.

Apparently it is the inverter rather than battery fw which introduces the change, reverting to 909 while staying on 3015 is reportedly possible and OK.

Thanks for looking at this!

gavanfantom, 2023-12-08:
I'm on battery 3013 and inverter 911. My fork of givenergy-modbus works on my setup, although I have to keep manually installing it with pip inside the docker container every time I update homeassistant.

cdpuk, 2023-12-13:
As per my comment on #61 I could do with some help testing a new release 2.0.0-alpha1 on newer hardware/firmware. This may or may not work at all, but sadly I have no way of knowing myself. Any feedback much appreciated.

btxluke, 2023-12-29:
Alpha Version 2 gives this error on attempted device setup:

Config flow could not be loaded: {"message":"Invalid handler specified"}

andynash, 2023-12-29:
> Alpha Version 2 gives this error on attempted device setup:
> 
> Config flow could not be loaded: {"message":"Invalid handler specified"}
> 
> Logger: homeassistant.config_entries
> Source: config_entries.py:2077
> First occurred: 09:24:30 (1 occurrences)
> Last logged: 09:24:30
> 
> Error occurred loading flow for integration givenergy_local: No module named 'givenergy_modbus.client.client'; 'givenergy_modbus.client' is not a package
> 
> 

FYI there's some more discussion on this under issue #61 that may help, though Alpha 2 is not yet fully working for me.

andynash, 2023-12-29:
FYI there's some more discussion on this under issue #61 that will help, though Alpha 2 is not yet fully working fully for me even after fixing this error.

btxluke, 2023-12-30:
Nice dice for me here - even with a manually updated modbus. I'll switch to GivTCP temporarily.

andynash, 2023-12-30:
> Nice dice for me here - even with a manually updated modbus. I'll switch to GivTCP temporarily.

Wish I could do that - I'm using HA Core, so no add-ons, and I haven't found any decent instructions for installing GivTCP and MTTQ manually - those I've found have just led me down rabbit holes, so I've had to give up.

One thing I did learn was that there is a lot more complexity (that I guess you don't see if you can install as add-ons, but it's there), and I really don't want to have to go that route if I can help it, even if I do manage to ever get it working.

btxluke, 2023-12-30:
> Wish I could do that - I'm using HA Core, so no add-ons, and I haven't found any decent instructions for installing GivTCP and MTTQ manually - those I've found have just led me down rabbit holes, so I've had to give up.

I'd say give HAOS a try - I'm running it on a dedicated raspberry PI 4, and it just works. I got givTCP & mosquitto set up in about 5 minutes flat.

andynash, 2023-12-30:
That's the long term plan, however migration is also an even bigger job with likely rabbit holes given the scale of what Im using HA for, so I can't go down that path unless I have plenty of time available - not the case at the moment sadly. I haven't even got as far as choosing hardware yet.

closed 2024-01-29 by cdpuk