cdpuk/givenergy-local #71: GivEnergy Local wont connect to Inverter

Piccyman, 2024-01-22:
Failed to connect to the inverter. The error appears when adding the device in devises and services.

I have a 13.5kW All In One and it has 2 IP addresses, one for the inverter/Battery and one for the gateway.
I have tried both and neither works.

GivEnergy Local update seems to be working ok

cdpuk, 2024-01-23:
This is strongly related to #61, which aims to add support for newer firmware versions found on non-AIO inverters. My understanding is that the AIO works slightly differently, so may require even more work. If you're able to run Home Assistant OS and want a quicker solution, you may want to try the giv-tcp project.

cdpuk, 2024-02-04:
Hi @Piccyman, while nothing has been done to explicitly support the AIO, v2.x of the integration communicates with devices in a very different way. I'd be curious to hear whether you see any change, and some logs would be interesting to see.
https://github.com/cdpuk/givenergy-local/blob/master/docs/troubleshooting.md#debug-logs

norfolkmustard, 2024-06-28:
Hi. Adding to this as I have the same situation. 

After enabling debug, seems there's still not much in the logs as it fails on initial onboarding of the inverter. I'm getting a very similar issue with GivTCP (can't find serial number but seems it can initially connect)

This error originated from a custom integration.

Logger: custom_components.givenergy_local
Source: custom_components/givenergy_local/config_flow.py:48
integration: GivEnergy Local (documentation, issues)
First occurred: 10:51:18 (1 occurrences)
Last logged: 10:51:18

Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 48, in async_step_user
    serial_no = await read_inverter_serial(user_input)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/config_flow.py", line 24, in read_inverter_serial
    await client.refresh_plant()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 126, in refresh_plant
    await self.execute(reqs, timeout=timeout, retries=retries)
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 313, in send_request_and_await_response
    raise asyncio.TimeoutError()
TimeoutError

This error originated from a custom integration.

Logger: custom_components.givenergy_local.givenergy_modbus.client.client
Source: custom_components/givenergy_local/givenergy_modbus/client/client.py:307
integration: givenergy_local (documentation, issues)
First occurred: 10:51:18 (22 occurrences)
Last logged: 10:55:56

Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x33 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60) after 1 tries at 1s, giving up

cdpuk, 2024-08-20:
An experimental branch for AIO support has now been created. Unless someone wants to sponsor me an AIO device, this is going to rely on a significant amount of trial & error with feedback from community testing. The first release is available as a HACS beta under tag `aio-dev-1`. This might not work in any meaningful way, however debug logs will help to start piece together the puzzle.

nnmalex, 2024-08-21:
With `aio-dev-1` I have:
Connecting to either AIO or GW:
```
This error originated from a custom integration.

Logger: custom_components.givenergy_local
Source: custom_components/givenergy_local/config_flow.py:48
integration: givenergy_local (documentation, issues)
First occurred: 11:09:11 (3 occurrences)
Last logged: 11:18:25

Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 48, in async_step_user
    serial_no = await read_inverter_serial(user_input)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/config_flow.py", line 24, in read_inverter_serial
    await client.refresh_plant()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 161, in refresh_plant
    await self.execute(reqs, timeout=timeout, retries=retries)
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 343, in send_request_and_await_response
    raise asyncio.TimeoutError()
TimeoutError
```
and:
```
This error originated from a custom integration.

Logger: custom_components.givenergy_local.givenergy_modbus.client.client
Source: custom_components/givenergy_local/givenergy_modbus/client/client.py:337
integration: givenergy_local (documentation, issues)
First occurred: 11:09:11 (41 occurrences)
Last logged: 11:20:35

Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x33 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60) after 1 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60) after 1 tries at 1s, giving up
```

I'm happy to debug if you tell me what to do.

cdpuk, 2024-08-21:
New experimental tag `aio-dev-2` is available for testing. Similar logs again would be useful. Thanks.

nnmalex, 2024-08-21:
‘aio-dev-2’ worked for me.

ARW-online, 2024-08-24:
Thank you!  `aio-dev-2` worked for me too, on an AIO running D0.611-A0.611

cdpuk, 2024-09-01:
Thanks for the feedback; that's very encouraging news. I've just published `aio-dev-3` with a tiny change to help guide further development. This is likely to either continue working as normal, or not work at all. In the latter case, just revert back to `aio-dev-2` to restore functionality.

Once I know whether this works, I can work on getting this merged in to the main release.

ARW-online, 2024-09-04:
Thanks!  I just tried `aio-dev-3`, and after the restart all the sensors went unavailable and it complained "Setup failed, will retry".  I rolled back to `aio-dev-2` and after the restart all the sensors returned as before.

cdpuk, 2024-09-07:
`aio-dev-4` is now ready. This should behave as per `aio-dev-2`, but also restores functionality for everyone else on Hybrid models.  Any reports of success or otherwise much appreciated.

ARW-online, 2024-09-07:
Thanks again! `aio-dev-4` seems to work as advertised with my AIO. Yay!

bootc, 2024-09-12:
I just wanted to report that I'm also using `aio-dev-4` successfully. I have _two_ AIOs in parallel mode, plus the requisite gateway, and an EV charger; only the AIOs can be added using this integration.

I do have an issue where every few minutes the integration goes _unavailable_ before coming back a little bit later; other than making everything go missing in HA this doesn't seem too problematic. The logs when this happens are:

```
2024-09-12 13:57:20.959 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0) after 3 tries at 1s, giving up
2024-09-12 13:57:20.959 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 13:57:20.960 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2024-09-12 13:57:21.209 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
2024-09-12 13:57:31.361 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Candidate frame found 136 bytes into buffer, discarding leading garbage: 0x434832343134473532320000003c00010000000010b8000009cb000000c400000000001900000000138c00050c2003e80000000000000000000000020000024e0002011e0000008202000017046e000001690000006b006a0043000000000000011f046b0252003d00000045000000150001794700d6028d09cb138c011f0097000000be0037bfea
2024-09-12 13:58:07.956 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0) after 3 tries at 1s, giving up
2024-09-12 13:58:07.957 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 13:58:07.957 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2024-09-12 13:58:08.208 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
2024-09-12 14:02:26.958 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0) after 3 tries at 1s, giving up
2024-09-12 14:02:26.958 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 14:02:26.958 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2024-09-12 14:02:27.211 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
2024-09-12 14:03:13.955 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0) after 3 tries at 1s, giving up
2024-09-12 14:03:13.956 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 14:03:13.958 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2024-09-12 14:03:14.208 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
2024-09-12 14:07:23.208 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
2024-09-12 14:07:23.208 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 14:07:23.208 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2024-09-12 14:07:33.355 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Candidate frame found 136 bytes into buffer, discarding leading garbage: 0x434832343134473532320000003c00010000000010b8000009c6000000c600000000001c00000000138b00050c1e03e80000000000000000000000020000029f0002011e000000820200000c04dc000001690000006b006a0044000000000000011d04d802a3003f00000047000000150001792f00f202e109c6138b011d0096000000c80036e36c
2024-09-12 14:12:28.958 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0) after 3 tries at 1s, giving up
2024-09-12 14:12:28.958 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 14:12:28.958 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
[... cut ...]
2024-09-12 14:13:20.955 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2024-09-12 14:13:20.956 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2024-09-12 14:13:21.205 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
```

cdpuk, 2024-09-15:
Thanks for the feedback, that's an interesting set of log messages due to the `discarding leading garbage` lines. This suggests the library doesn't recognise a response from the inverter, so anything after that point might be difficult to interpret correctly, causing a disconnect. Some time will be needed to decode and understand those responses.

Is anyone else with an AIO seeing log messages like that? Knowing whether this applies only to certain models/generations is also useful.

norfolkmustard, 2024-09-15:
Hi. seeing a couple of timeouts but not the same "leading garbage"

`2024-09-15 22:03:03.076 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 10.10.20.42 (attempt=1/3, full_refresh=False)
2024-09-15 22:03:03.077 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=0)
2024-09-15 22:03:03.329 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=180)
2024-09-15 22:03:04.079 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0), attempting retry 1 of 2
2024-09-15 22:03:04.080 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 2): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=0)
2024-09-15 22:03:04.330 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180), attempting retry 1 of 2
2024-09-15 22:03:04.332 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 2): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=180)
2024-09-15 22:03:05.082 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0), attempting retry 2 of 2
2024-09-15 22:03:05.084 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 3): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=0)
2024-09-15 22:03:05.335 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180), attempting retry 2 of 2
2024-09-15 22:03:05.337 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 3): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=180)

2024-09-15 22:03:06.086 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0) after 3 tries at 1s, giving up
2024-09-15 22:03:06.087 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error:
2024-09-15 22:03:06.088 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-09-15 22:03:06.092 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error

2024-09-15 22:03:06.092 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 3.017 seconds (success: False)
2024-09-15 22:03:06.103 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2024-09-16 00:30:00
2024-09-15 22:03:06.109 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2024-09-16 00:01:00
2024-09-15 22:03:06.338 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180) after 3 tries at 1s, giving up
2024-09-15 22:03:16.080 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 10.10.20.42:8899
2024-09-15 22:03:16.080 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Performing model detection`

also seeing this occasionally

`2024-09-15 21:43:46.083 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 10.10.20.42 (attempt=1/3, full_refresh=False)
2024-09-15 21:43:46.088 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=0)
2024-09-15 21:43:46.239 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-09-15 21:43:46.240 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0)
2024-09-15 21:43:46.241 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0)
2024-09-15 21:43:46.340 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x11 base_register=180)
2024-09-15 21:43:46.496 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-09-15 21:43:46.497 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=180)
[... cut ...]
2024-09-15 21:43:53.797 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-09-15 21:43:53.798 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0)
2024-09-15 21:43:53.798 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0)`

cdpuk, 2024-11-09:
This has been merged in with the v2.1.0 release.  This release bundles in a few other fixes and features that may or may not improve the AIO experience, in particular:
* A much upgraded debugging script (see the docs) that will be useful to further investigate issues.
* CRC checking on received data.

Please raise new issues for any specific things that need further investigation.

closed 2024-11-09 by cdpuk