cdpuk/givenergy-local #75: Stopped working since yesterday's upgrade

TimWardCam, 2024-01-30:
## Hardware and versions

* Inverter model & generation:

AC Gen 1 3.0kW

* Inverter firmware version:

D0.535-A0.535

* Battery firmware version:

3013

* Home Assistant integration version:

I can't see where to find that. It's the latest version that was updated to yesterday.

## Describe the bug

Since the integration was updated yesterday it's stopped working. All sensors show Unavailable and the overnight automations didn't run.

## Debug log

```text

Add your logs here.

2024-01-30 08:25:20.469 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.14:8899
2024-01-30 08:25:20.469 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.14 (attempt=1/3, full_refresh=True)
2024-01-30 08:25:20.471 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 08:25:20.474 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-01-30 08:25:20.474 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.032 seconds (success: False)
2024-01-30 08:25:30.465 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.14:8899
2024-01-30 08:25:30.466 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.14 (attempt=1/3, full_refresh=True)
2024-01-30 08:25:30.468 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 08:25:30.472 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-01-30 08:25:30.472 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.030 seconds (success: False)
2024-01-30 08:25:40.466 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.14:8899
2024-01-30 08:25:40.466 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.14 (attempt=1/3, full_refresh=True)
2024-01-30 08:25:40.468 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 08:25:40.470 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-01-30 08:25:40.471 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.028 seconds (success: False)
2024-01-30 08:25:50.466 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.14:8899
2024-01-30 08:25:50.466 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.14 (attempt=1/3, full_refresh=True)
2024-01-30 08:25:50.468 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 08:25:50.469 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-01-30 08:25:50.470 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.028 seconds (success: False)
2024-01-30 08:26:00.549 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.14:8899
2024-01-30 08:26:00.549 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.14 (attempt=1/3, full_refresh=True)
2024-01-30 08:26:00.550 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 08:26:00.555 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-01-30 08:26:00.555 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.029 seconds (success: False)
2024-01-30 08:26:10.466 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.14:8899
2024-01-30 08:26:10.466 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.14 (attempt=1/3, full_refresh=True)
2024-01-30 08:26:10.468 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 08:26:10.473 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-01-30 08:26:10.473 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.031 seconds (success: False)

alex250472, 2024-01-30:
I am having same issue as well, although i was on previous version 1.9.1. Stopped working at midnight last night. Have tried removing and reinstalling, newer and older versions but all have same issue. Also powered down/up router and inverter, but still same.

Inverter Model:
AC Gen 1 3.0kW

Software version:
D0.535-A0.535

2024-01-30 10:13:10.689 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x36 base_register=60)
2024-01-30 10:13:10.924 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-01-30 10:13:10.924 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60)
2024-01-30 10:13:10.925 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60)
2024-01-30 10:13:10.934 WARNING (MainThread) [custom_components.givenergy_local.coordinator] Data discarded: temp_charger value of 6542.8 is out of range [-10, 100]
2024-01-30 10:13:10.934 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 8.285 seconds (success: False)
2024-01-30 10:13:10.935 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Shutting down
2024-01-30 10:13:10.935 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-01-30 10:13:10.937 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection

cdpuk, 2024-01-30:
Thanks for the feedback. Unfortunately nobody tested v2.0.0 with an AC inverter.

@alex250472 you mention it stopped at midnight. Did v2.0.0 actually run successfully for some period before this? The fact you've tried going back to 1.9.1 and that doesn't work sounds very strange. I've never even seen a case of having to restart an inverter before.

@TimWardCam did yours work at all? Or did it also stop at midnight?

In any case, v2.0.1 has just been published. This includes some additional logging that might help trace what's going on. Please give this a try and capture some new logs. The line we're looking for is "Closing connection due to expected error", with hopefully some further context after.

alex250472, 2024-01-30:
Was on 1.91 when graph history flat lined at midnight last night. I upgraded to 2.0 this morning to see if this would fix.

Debug log attached after 2.01 was installed.
[home-assistant_givenergy_local_2024-01-30T21-18-09.767Z.log](https://github.com/cdpuk/givenergy-local/files/14104463/home-assistant_givenergy_local_2024-01-30T21-18-09.767Z.log)

TimWardCam, 2024-01-30:
Mine might have stopped at midnight too. The last readings were just before. So it was probably midnight that did it rather than the time of the upgrade.

TimWardCam, 2024-01-30:
I'm not being offered another upgrade. How do I find out what version I'm running?

alex250472, 2024-01-30:
I was upgraded to 2.0 when stopped working. Sorry for confusion.

TimWardCam, 2024-01-30:
I've found the version number I'm running - it's v2.0.1

TimWardCam, 2024-01-30:
I've just been offered v2.01. I'll upgrade and capture the debug log.

TimWardCam, 2024-01-31:
But overnight there appears to be a gap in data collected between 0058 and 0128, and although HA said that the overnight cheap rate charging automation worked in fact it looks like it didn't do the right thing. There are various other automations to run today - I'll keep an eye on them.

ianfretwell, 2024-01-31:
Updated to 2.0.1 from 1.9.1 this morning.

Inverter 'disappeared' for approx 10 minutes.  Several reloads of the integration and it came back.

It's since done this several more times since - bear in mind that everything has been fine on 1.9.1.

It's as if it's crashing the wifi dongle somehow - I can still see the dongle connected to my router,  but its not pingable.

AC3   Inverter D0.203-A0.203 / Battery 3315

Also - when it is working - I've noticed that the GiveEnergy app, when connected as 'Home' is only refreshing every 32 seconds or so rather than the usual 10.

Gut feeling is that the change to 10 second polling is overloading something on the AC3's.

cdpuk, 2024-01-31:
You could be right about the update rate. It's completely solid on my 5.0 Gen1 @ 10s, with Wi-Fi dongle, but that doesn't mean we don't have issues with other kit.

The midnight thing is really strange. I'm going to park that thought until a better explanation can be found. It almost sounds like a firmware error due to it being day 31 of the month!

If you want to try changing the update rate, you need `coordinator.py` line 73:
```
update_interval=timedelta(seconds=10)
```  
Try setting that back to 30 and let us know if that makes any difference.

ianfretwell, 2024-02-01:
It ran for alomost 24 hours without an issue whilst still on 10...so I'm trying with 15 now just to see if that little bit is enough.

Will be back with an update as soon as I know.

TimWardCam, 2024-02-02:
But since the incident the problem I have now is that regardless of what I tell the battery to charge up to on a timed charge it now always charges up to 100%. This is not helpful, particularly for the afternoon top-up which is only supposed to charge it up enough, if necessary, to get through the 1600 - 1900 expensive period of the tariff.

ianfretwell, 2024-02-02:
Already failed at 20 seconds.  Logs show some entries that tie up with those being posted on the other issue thread:-

Error fetching Inverter data: Connection closed due to expected error
10:27:31 – (ERROR) GivEnergy Local (custom integration) - message first occurred at 10:11:17 and shows up 2 times

Inverter model failed validation: 257 is not a valid UsbDevice
10:27:31 – (WARNING) GivEnergy Local (custom integration)

What kind of error is an 'expected' one ?

ianfretwell, 2024-02-02:
Well - I'm now pretty sure it's NOT the time interval being lower than 30.

I set it to that earlier and it lasted just over 5 hours before failing again. So it looks like the logs errors (e.g. 'not a valid USBdevice') are more likely the cause.

TimWardCam, 2024-02-02:
My system just lost contact for 21 minutes again, and the wrong things were programmed into the inverter (or the right things didn't get programmed during the outage), and I had to sort it out manually.

How can I revert to the previously working version of the integration?

TimWardCam, 2024-02-03:
It's still broken.

I only installed Home Assistant so as to be able to control the Givenergy kit, and I'm now having to control it manually.

PLEASE ... how do I "down"grade to an earlier working version?

ianfretwell, 2024-02-03:
Lol...you installed HA purely for GivEnergy?...it'd have been easier to just use the app manually!

Anyway - assuming you installed through HACS: -

Go to HACS, then to your integrations, open GivEnergy, click the three little dots top right and select Redownload - then pick the version you want (v1.9.1)...restart HA. Done.

TimWardCam, 2024-02-03:
I didn't want to fiddle with the app at least twice a day - zero times for preference. Specifically I wanted to be able to charge overnight during the cheap period of the tariff, and top up if/as necessary just before the expensive period. Automating the "free electricity" sessions is a plus, as is automating the energy saving sessions particularly as Octopus are utterly crap at giving notice of them other than via the API. 

Thanks for pointing me at the well-hidden menu option. I'll try that.

cdpuk, 2024-02-03:
v2.0.2 is now available, which should improve things, in particular for those seeing "not a valid UsbDevice" in their logs.

TimWardCam, 2024-02-03:
I'm now back with 1.9.1 and both problems have gone away.

(1) The "nothing works for a while" connectivity issue.

(2) The Enable timed charging issue.

I realise that I probably haven't explained (2) properly. I have automations that want to top the battery up to a defined level 
[image]
With version 2.x it was always charging to 100% whatever level was set. Back on 1.9.1 it appears to have stopped doing that.

alex250472, 2024-02-03:
I’m also back on 1.9.1 and all problems gone away

cdpuk, 2024-02-04:
I'm closing this one down as I'm not convinced that this thread is AC specific and distinct to the issues being discussed in #76.

I'm also unable to help if everyone simply downgrades to the last known good version.

Please contribute to #76 if you're having intermittent connectivity issues, or raise new issues for anything specific such as charge targets not working.

closed 2024-02-04 by cdpuk