cdpuk/givenergy-local #76: Givenergy entities dropping out

DominicMcCann, 2024-01-31:
## Hardware and versions

* Inverter model & generation:
* Inverter firmware version:
* Battery firmware version:
* Home Assistant integration version:

## Describe the bug

A clear and concise description of what the bug is.

## Debug log

```text

Add your logs here.

```

DominicMcCann, 2024-01-31:
My Givenergy integration is now reporting “entities not available”.
Is anyone else getting this?
The integration was updated about 1 or 2 days ago but I am not sure exactly how long the entities have stopped being available.
I did restart HA and then the Givenergy entities re-appeared for a couple of hours but then stopped working again.
Does any one have any ideas how to troubleshoot this kind of issue with the Givenergy integration?

ianfretwell, 2024-01-31:
Same here - posting more detail on the other issue thread...

smoothquark, 2024-01-31:
Yep, same issue. Problem started after latest update. Restarting the integration gets them up and running again.

cdpuk, 2024-01-31:
As per the AC inverter thread, the current line of thinking is to reduce the update rate back to 30s. It was changed to 10s in 2.0.0 since this was found to be stable for those who tested it.

If you want to try changing the update rate, you need `givenergy_local/coordinator.py` line 73:
```
update_interval=timedelta(seconds=10)
```
You can try dropping 30 back in there and restart HA.

If that doesn't help, some debug logs would be needed to trace this further.

DominicMcCann, 2024-02-02:
Yeah I haven't found how to change the interval parameter and the Givenergy entities are still constantly dropping out.
Here are some error logs:
Error fetching Inverter data: Error communicating with inverter:
1 February 2024 at 17:15:22 – (ERROR) GivEnergy Local (custom integration) - message first occurred at 31 January 2024 at 12:58:23 and shows up 101 times
Inverter model failed validation: 257 is not a valid UsbDevice
1 February 2024 at 17:15:22 – (WARNING) GivEnergy Local (custom integration) - message first occurred at 31 January 2024 at 13:41:55 and shows up 4 times
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
1 February 2024 at 17:13:41 – (WARNING) GivEnergy Local (custom integration) - message first occurred at 31 January 2024 at 12:58:23 and shows up 293 times
Error doing job: Task exception was never retrieved
1 February 2024 at 16:55:37 – (ERROR) GivEnergy Local (custom integration) - message first occurred at 1 February 2024 at 16:53:46 and shows up 12 times

ianfretwell, 2024-02-02:
Eh ?    It literally TELLS YOU HOW TO CHANGE THE INTERVAL PARAMETER in the post you just replied to!!  

_If you want to try changing the update rate, you need givenergy_local/coordinator.py line 73:

update_interval=timedelta(seconds=10)
You can try dropping 30 back in there and restart HA._

DominicMcCann, 2024-02-02:
Here is some more debug stuff:
2024-01-31 12:56:11.044 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration openevse which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.046 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration givenergy_local which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.051 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.054 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration alarmo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.056 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration zha_toolkit which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.059 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration google_home which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.062 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration octopus_energy which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.064 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration nibe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:11.067 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration mass which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-31 12:56:35.550 ERROR (SyncWorker_4) [homeassistant.components.influxdb] InfluxDB bucket is not accessible due to '(401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.7.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Wed, 31 Jan 2024 12:56:35 GMT', 'Content-Length': '55'})
HTTP response body: {"code":"unauthorized","message":"unauthorized access"}
'. Please check that the bucket, org and token are correct and that the token has the correct permissions set. Retrying in 60 seconds.
2024-01-31 12:56:36.222 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Unexpected error fetching Inverter data: Error connecting to 192.168.1.135:8899
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1112, in create_connection
    transport, protocol = await self._create_connection_transport(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1145, in _create_connection_transport
    await waiter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 61, in connect
    self.reader, self.writer = await asyncio.wait_for(
                               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 300, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/coordinator.py", line 90, in _async_update_data
    await self.client.connect()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 65, in connect
    raise CommunicationError(
custom_components.givenergy_local.givenergy_modbus.exceptions.CommunicationError: Error connecting to 192.168.1.135:8899
2024-01-31 12:56:49.425 ERROR (MainThread) [homeassistant.components.media_player] Error while setting up vlc platform for media_player
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
[... cut ...]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/vlc/media_player.py", line 48, in setup_platform
    [VlcDevice(config.get(CONF_NAME, DEFAULT_NAME), config.get(CONF_ARGUMENTS))]
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/vlc/media_player.py", line 68, in __init__
    self._instance = vlc.Instance(arguments)
                     ^^^^^^^^^^^^^^^^^^^^^^^

smoothquark, 2024-02-02:
> As per the AC inverter thread, the current line of thinking is to reduce the update rate back to 30s. It was changed to 10s in 2.0.0 since this was found to be stable for those who tested it.
>
> If [...]

Changed to 30 seconds from 10 seconds and has been stable so far for more than 12 hours. Previously, with 10 seconds, the entities would drop out within an hour or so after a restart.

smoothquark, 2024-02-03:
Hmm... I spoke too soon. Dropped out at some point last night. Reload has got it up and running again. Here are the (hopefully) relevant logs...

Unexpected error fetching Inverter data: Error connecting to 192.168.86.175:8899
05:13:10 – (ERROR) GivEnergy Local (custom integration)
Error doing job: Task exception was never retrieved
00:05:00 – (ERROR) GivEnergy Local (custom integration) - message first occurred at 2 February 2024 at 23:30:00 and shows up 3 times
Inverter model failed validation: hour must be in 0..23
2 February 2024 at 22:11:08 – (WARNING) GivEnergy Local (custom integration) - message first occurred at 2 February 2024 at 08:54:23 and shows up 4 times
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2 February 2024 at 22:10:36 – (WARNING) GivEnergy Local (custom integration) - message first occurred at 2 February 2024 at 09:56:24 and shows up 6 times
Error fetching Inverter data: Error communicating with inverter:
2 February 2024 at 22:10:35 – (ERROR) GivEnergy Local (custom integration) - message first occurred at 2 February 2024 at 09:56:24 and shows up 2 times
Error doing job: Task exception was never retrieved
2 February 2024 at 21:22:13 – (ERROR) GivEnergy Local (custom integration) - message first occurred at 2 February 2024 at 21:20:13 and shows up 5 times
Data discarded: temp_inverter_heatsink value of 4932.0 is out of range [-10, 100]
2 February 2024 at 14:37:57 – (WARNING) GivEnergy Local (custom integration) - message first occurred at 1 February 2024 at 20:01:55 and shows up 4 times
Error doing job: Unclosed client session
2 February 2024 at 14:32:26 – (ERROR) runner.py - message first occurred at 2 February 2024 at 11:18:41 and shows up 5 times
Updating state for sensor.pv_energy_today (<class 'custom_components.givenergy_local.sensor.PVEnergyTodaySensor'>) took 0.683 seconds. Please create a bug report at https://github.com/cdpuk/givenergy-local/issues
2 February 2024 at 14:32:26 – (WARNING) helpers/entity.py - message first occurred at 2 February 2024 at 08:53:52 and shows up 2 times
Entity sensor.consumption_today from integration givenergy_local has state class total_increasing, but its state is not strictly increasing. Triggered by state 22.8 (22.9) with last_updated set to 2024-02-01T15:54:51.685970+00:00. Please create a bug report at https://github.com/cdpuk/givenergy-local/issues

I think it dropped out at around 22:00 last night.

DominicMcCann, 2024-02-03:
Yeah, I have downloaded update from yesterday v2.0.1 and it is possibly a little more stable, but still experiencing dropping out. I have done a couple of re-loads /re-starts of the integration and it does bring it but, but it usually drops out again after a couple of hours. I will keep watching the behaviour and post anther update probably tomorrow.

cdpuk, 2024-02-03:
v2.0.2 is now available, which should improve things, in particular for those seeing "not a valid UsbDevice" in their logs.

The refresh interval remains at 10s for now. If there are still problem after this update, I might back this off a bit. Just trying to change one thing at a time.

For those posting logs, please remember to include full debug logs, not just those you see in the HA UI. The UI logs only show us what's broken, but they don't explain the events that led up to the broken state.
https://github.com/cdpuk/givenergy-local/blob/1d0b3f35208c2b74648d0c4ceaeb8a803506f8a4/.github/ISSUE_TEMPLATE/issue.md?plain=1#L36

DominicMcCann, 2024-02-04:
Yeah I'm still only getting an hour or so and then the Entities drop out again.

DominicMcCann, 2024-02-04:
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.7.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Sun, 04 Feb 2024 10:28:17 GMT', 'Content-Length': '55'})
HTTP response body: {"code":"unauthorized","message":"unauthorized access"}
'. Please check that the bucket, org and token are correct and that the token has the correct permissions set. Retrying in 60 seconds.
2024-02-04 10:28:21.722 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:28:31.721 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:28:41.722 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:28:51.721 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:29:01.722 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:29:11.720 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:29:17.608 ERROR (SyncWorker_10) [homeassistant.components.influxdb] InfluxDB bucket is not accessible due to '(401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.7.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Sun, 04 Feb 2024 10:29:17 GMT', 'Content-Length': '55'})
HTTP response body: {"code":"unauthorized","message":"unauthorized access"}
'. Please check that the bucket, org and token are correct and that the token has the correct permissions set. Retrying in 60 seconds.
2024-02-04 10:29:21.722 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:29:31.722 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:29:41.723 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:29:51.722 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:30:01.725 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:30:11.721 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:30:17.624 ERROR (SyncWorker_2) [homeassistant.components.influxdb] InfluxDB bucket is not accessible due to '(401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.7.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Sun, 04 Feb 2024 10:30:17 GMT', 'Content-Length': '55'})
HTTP response body: {"code":"unauthorized","message":"unauthorized access"}
'. Please check that the bucket, org and token are correct and that the token has the correct permissions set. Retrying in 60 seconds.
2024-02-04 10:30:21.724 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:30:31.724 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:30:41.725 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:30:51.724 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:31:01.724 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:31:11.729 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:31:17.640 ERROR (SyncWorker_5) [homeassistant.components.influxdb] InfluxDB bucket is not accessible due to '(401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.7.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Sun, 04 Feb 2024 10:31:17 GMT', 'Content-Length': '55'})
HTTP response body: {"code":"unauthorized","message":"unauthorized access"}
'. Please check that the bucket, org and token are correct and that the token has the correct permissions set. Retrying in 60 seconds.
2024-02-04 10:31:21.734 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
[... cut ...]
2024-02-04 10:50:42.721 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 257 is not a valid UsbDevice
2024-02-04 10:50:42.721 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-02-04 10:50:42.726 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
2024-02-04 10:50:42.726 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.015 seconds (success: False)

cdpuk, 2024-02-04:
@DominicMcCann have you updated to v2.0.2 and restarted HA? Those logs look a lot like the errors from v2.0.1.

ianfretwell, 2024-02-04:
28 hours of uptime since moving to 2.0.2 - no issues so far.

chrisdwest1984, 2024-02-04:
I'm afraid I'm getting issues still on V2.0.2 with an AC Coupled Inverter. It doesn't connect at all and appears as 'Unavailable' in HA. Running fine on 1.9.1. :

Logger: homeassistant.config_entries
Source: config_entries.py:442
First occurred: 12:27:17 PM (1 occurrences)
Last logged: 12:27:17 PM

Error setting up entry Solar Inverter (S/N <removed>) for givenergy_local
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 442, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/__init__.py", line 26, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 278, in async_config_entry_first_refresh
    await self._async_refresh(
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 313, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/coordinator.py", line 132, in _async_update_data
    await asyncio.sleep(_REFRESH_DELAY_BETWEEN_ATTEMPTS)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 665, in sleep
    return await future
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

ianfretwell, 2024-02-04:
Could issues now be firmware version related since v2 was rewitten for newer firmware versions ?

i.e. "Compatibility with new inverter models and updated firmware."

My AC is on D0.203-A0.203 (so the latest beta version).

smoothquark, 2024-02-04:
Mine has been stable here for more than 24 hours since updating to 2.0.2. Fingers crossed.

chrisdwest1984, 2024-02-04:
My AC is on D0.535-A0.535

DominicMcCann, 2024-02-04:
Sorry to be a total novice here, but if HA isn't offering to automatically update from v2.0.1 to v2.0.2 is there a way to force it to update. Do I need to uninstall / re-install? Will I need to re-add my login credentials etc?

cdpuk, 2024-02-04:
@DominicMcCann if HACS isn't seeing new versions, it's probably because it only checks so often. You can trick it in to checking again by browsing to the givenergy_local integration via HACS, then go back, and it should show any upgrades available. You won't need to re-enter any credentials when switching versions via HACS.

@chrisdwest1984 please can you raise a new issue for the issues you're seeing. I'm trying to keep this thread focused on stability issues. If you can't connect at all, that's something different. I'll need full debug logs to understand what's going on - you'll see instructions on the create issue page on GitHub.

DominicMcCann, 2024-02-05:
OK it seems to be much more stable now with v2.0.2. 
My Givenergy data has been coming through continuously since my last update ~17:00hr yesterday (Sunday 04/02/2024) afternoon.

cdpuk, 2024-02-06:
Sounds like things are good enough for now, and most of the issues seen with the introduction of v2.0.0 have been sorted. If there's anything else, please raise a new issue so we can address specific problems.

closed 2024-02-06 by cdpuk