cdpuk/givenergy-local #78: AC Coupled inverter unavailable in V2.0.x

chrisdwest1984, 2024-02-04:
## Hardware and versions

* Inverter model & generation: AC Coupled
* Inverter firmware version: D0.535-A0.535
* Battery firmware version: 3015
* Home Assistant integration version: 2.0.2

## Describe the bug

When upgrading the integration to V2.0.x (all versions of V2) GivEnergy entities all show as 'unavailable'

## Debug log

```text

2024-02-04 12:25:13.567 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(TransparentRequest)
2024-02-04 12:25:13.567 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(TransparentResponse)
2024-02-04 12:25:13.567 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(NullResponse)
2024-02-04 12:25:13.575 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadRegistersMessage)
2024-02-04 12:25:13.575 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadRegistersRequest)
2024-02-04 12:25:13.580 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadRegistersResponse)
2024-02-04 12:25:13.580 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadHoldingRegisters)
2024-02-04 12:25:13.580 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadHoldingRegistersRequest)
2024-02-04 12:25:13.580 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadHoldingRegistersResponse)
2024-02-04 12:25:13.580 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadInputRegisters)
2024-02-04 12:25:13.581 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadInputRegistersRequest)
2024-02-04 12:25:13.581 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadInputRegistersResponse)
2024-02-04 12:25:13.581 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadBatteryInputRegisters)
2024-02-04 12:25:13.581 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadBatteryInputRegistersRequest)
2024-02-04 12:25:13.581 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadBatteryInputRegistersResponse)
2024-02-04 12:25:13.588 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(WriteHoldingRegister)
2024-02-04 12:25:13.588 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(WriteHoldingRegisterRequest)
2024-02-04 12:25:13.589 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(WriteHoldingRegisterResponse)

2024-02-04 12:25:20.923 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Unexpected error fetching Inverter data: Error connecting to 192.168.68.148:8899
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1146, in create_connection
    transport, protocol = await self._create_connection_transport(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1179, in _create_connection_transport
    await waiter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 61, in connect
    self.reader, self.writer = await asyncio.wait_for(
                               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 313, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/coordinator.py", line 92, in _async_update_data
    await self.client.connect()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 65, in connect
[... 228 lines cut ...]
2024-02-04 12:26:14.460 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-02-04 12:26:14.460 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60)
2024-02-04 12:26:14.461 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60)
2024-02-04 12:26:14.461 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] First time encountering slave address 0x34
2024-02-04 12:26:14.481 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x35 base_register=60)
2024-02-04 12:26:14.710 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-02-04 12:26:14.710 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60)
2024-02-04 12:26:14.710 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60)
2024-02-04 12:26:14.711 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] First time encountering slave address 0x35
2024-02-04 12:26:14.732 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x36 base_register=60)
2024-02-04 12:26:14.965 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-02-04 12:26:14.965 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60)
2024-02-04 12:26:14.965 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60)
2024-02-04 12:26:14.966 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] First time encountering slave address 0x36
2024-02-04 12:26:14.969 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Updating connected battery count to 1
2024-02-04 12:26:14.971 WARNING (MainThread) [custom_components.givenergy_local.coordinator] Inverter model failed validation: minute must be in 0..59
2024-02-04 12:26:16.160 WARNING (MainThread) [custom_components.localtuya.common] [bf1...zfb] Failed to connect to 192.168.68.142: [Errno 113] Connect call failed ('192.168.68.142', 6668)
[... cut ...]
2024-02-04 12:26:17.422 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-02-04 12:26:17.423 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)
2024-02-04 12:26:17.424 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)

```

cdpuk, 2024-02-04:
Thanks for the detailed feedback. Please can you capture the same logs against v2.0.3? This will provide some more information in the "Inverter model failed validation" message. Looks like one of the date/time properties isn't as expected on AC devices.

chrisdwest1984, 2024-02-04:
Here you go - thanks!:

```
2024-02-04 15:35:36.555 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(TransparentRequest)
2024-02-04 15:35:36.555 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(TransparentResponse)
2024-02-04 15:35:36.556 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(NullResponse)
2024-02-04 15:35:36.557 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadRegistersMessage)
2024-02-04 15:35:36.558 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadRegistersRequest)
2024-02-04 15:35:36.558 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadRegistersResponse)
2024-02-04 15:35:36.558 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadHoldingRegisters)
2024-02-04 15:35:36.558 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadHoldingRegistersRequest)
2024-02-04 15:35:36.558 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadHoldingRegistersResponse)
2024-02-04 15:35:36.558 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadInputRegisters)
2024-02-04 15:35:36.559 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadInputRegistersRequest)
2024-02-04 15:35:36.559 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadInputRegistersResponse)
2024-02-04 15:35:36.559 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadBatteryInputRegisters)
2024-02-04 15:35:36.559 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadBatteryInputRegistersRequest)
2024-02-04 15:35:36.559 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(ReadBatteryInputRegistersResponse)
2024-02-04 15:35:36.562 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(WriteHoldingRegister)
2024-02-04 15:35:36.562 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(WriteHoldingRegisterRequest)
2024-02-04 15:35:36.562 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.pdu.transparent] TransparentMessage.__init_subclass__(WriteHoldingRegisterResponse)

2024-02-04 15:35:40.902 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Unexpected error fetching Inverter data: Error connecting to 192.168.68.148:8899
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1146, in create_connection
    transport, protocol = await self._create_connection_transport(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1179, in _create_connection_transport
    await waiter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 61, in connect
    self.reader, self.writer = await asyncio.wait_for(
                               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 313, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/coordinator.py", line 92, in _async_update_data
    await self.client.connect()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 65, in connect
    raise CommunicationError(
custom_components.givenergy_local.givenergy_modbus.exceptions.CommunicationError: Error connecting to 192.168.68.148:8899
2024-02-04 15:35:40.913 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 3.963 seconds (success: False)
[... cut ...]
2024-02-04 15:36:47.669 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 15.431 seconds (success: False)
2024-02-04 15:36:47.670 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Shutting down
2024-02-04 15:36:47.670 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-02-04 15:36:47.674 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection
```

cdpuk, 2024-02-04:
So that's saying that the 2nd discharge time slot can't be interpreted as expected. How does that currently appear in the GE app or web portal?

chrisdwest1984, 2024-02-04:
I have Discharge Mode set to 'Dynamic'.

However, when I change it to 'Scheduled' the start and end time are both 00:00 .

cdpuk, 2024-02-04:
Ah my bad. I forgot the app only shows you one set of times, but there's normally additional schedules that can be seen via https://www.givenergy.cloud/. If you go there, then "My Inverter", then the Remote Control icon in the top right, you should see a more detailed section for "Charge/Discharge Schedules".

chrisdwest1984, 2024-02-04:
Ah right!

[image]

It looks like this. Presume I should change the second one so it displays a time?

cdpuk, 2024-02-04:
Yeah. It's interesting that it's blank by default - the likely cause of the problem. Perhaps if you set a time, things will jump to life? Or at least show a different error 😄

chrisdwest1984, 2024-02-04:
I set a time of 00:00 and that seems to have sorted it with v2.0.3. Thanks very much for your help!

cdpuk, 2024-02-04:
Great. I don't think it's right for the integration to try and fix/hide these values, so I've updated the docs to cover this. Hopefully it's just a one-off fix, but please report back if not.

https://github.com/cdpuk/givenergy-local/blob/master/docs/troubleshooting.md#timeslots-with-invalid-values

closed 2024-02-04 by cdpuk