cdpuk/givenergy-local #81: Failed to setup - will retry

IRH1980, 2024-04-07:
* Inverter GIV-HY5.0 Gen 3
* Inverter firmware version: D0.307-A0.307
* Battery firmware version: 3015
* Home Assistant integration version: 2.0.3

I had the integration working fine, but i had to speak with Givenergy and i think they pushed an updated of some sort to the equipment; since then i can’t get a connection. I can telnet into the IP and port, getting the blinking cursor and eventually some scrolling characters, but the integration gives me ‘no devices or entities’ and ‘failed to setup, will retry’. I have removed it and started again but still no joy.

Debug log

```text

2024-04-07 23:43:12.517 INFO (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Connection established to 192.168.1.121:8899
2024-04-07 23:43:12.517 INFO (MainThread) [custom_components.givenergy_local.coordinator] Fetching data from 192.168.1.121 (attempt=1/3, full_refresh=True)
2024-04-07 23:43:12.518 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x32 base_register=0)
2024-04-07 23:43:12.760 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:12.761 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)
2024-04-07 23:43:12.762 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)
2024-04-07 23:43:12.768 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x32 base_register=180)
2024-04-07 23:43:13.000 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:13.000 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=180)
2024-04-07 23:43:13.000 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=180)
2024-04-07 23:43:13.019 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:3/ReadHoldingRegistersRequest(slave_address=0x32 base_register=0)
2024-04-07 23:43:13.255 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:13.256 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=0)
2024-04-07 23:43:13.257 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=0)
2024-04-07 23:43:13.272 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:3/ReadHoldingRegistersRequest(slave_address=0x32 base_register=60)
2024-04-07 23:43:13.510 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:13.510 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=60)
2024-04-07 23:43:13.510 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=60)
2024-04-07 23:43:13.522 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:3/ReadHoldingRegistersRequest(slave_address=0x32 base_register=120)
2024-04-07 23:43:13.752 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:13.752 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=120)
2024-04-07 23:43:13.753 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=120)
2024-04-07 23:43:13.773 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x32 base_register=120)
2024-04-07 23:43:14.000 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:14.001 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=120)
2024-04-07 23:43:14.001 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=120)
2024-04-07 23:43:14.024 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x32 base_register=60)
2024-04-07 23:43:14.261 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:14.262 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60)
2024-04-07 23:43:14.263 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60)
2024-04-07 23:43:14.275 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x33 base_register=60)
2024-04-07 23:43:14.500 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:14.501 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x33 base_register=60)
2024-04-07 23:43:14.501 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x33 base_register=60)
2024-04-07 23:43:14.501 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] First time encountering slave address 0x33
2024-04-07 23:43:14.526 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x34 base_register=60)
2024-04-07 23:43:14.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:14.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60)
2024-04-07 23:43:14.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] Handling 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60)
2024-04-07 23:43:14.751 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.model.plant] First time encountering slave address 0x34
2024-04-07 23:43:14.777 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Request sent (attempt 1): 2:4/ReadInputRegistersRequest(slave_address=0x35 base_register=60)
2024-04-07 23:43:15.000 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.framer] Found next frame: 0x59590001009e0102..., buffer_len=164
2024-04-07 23:43:15.001 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Processing 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60)
[... cut ...]
2024-04-07 23:43:27.007 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 14.492 seconds (success: False)
2024-04-07 23:43:27.008 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Shutting down
2024-04-07 23:43:27.009 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Disconnecting and cleaning up
2024-04-07 23:43:27.013 DEBUG (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] network_consumer reader at EOF, cannot continue, closing connection

```

cdpuk, 2024-04-27:
This looks like another GivEnergy update that breaks compatibility. I see you've got giv_tcp working, so we can look to port whatever change was required over to here at some point.

IRH1980, 2024-04-27:
Yeah givtcp seems to have been ok since I loaded it up. Hope it all ports over ok!

Lmcde3, 2024-05-21:
+1 for keeping this integration alive. I use it a lot as I find GivTCP does not allow for some entities

cdpuk, 2024-11-09:
Interested to hear whether anything has changed in the recent v2.1.0 release - there have been a lot of changes so in any case a new set of logs would be useful to troubleshoot this further.

cdpuk, 2025-01-17:
Issue appears to be stale

closed 2025-01-17 by cdpuk