cdpuk/givenergy-local #82: Integration "Failed Setup" - currently bricked

rongalbraith, 2024-05-16:
## Hardware and versions
* Inverter model & generation: GIV-AC3.0
* Inverter firmware version: D0.535-A0.535
* Battery firmware version: 3015
* Home Assistant integration version: 2.0.3

## Describe the bug
Overnight the integration failed.  It's been running for some time without issue but now it can't see the devices/entities.
Tried a restart and reload but no luck, grabbed some log files and hopefully this will help determine the problem (looks like the Givenergy data context may have changed).

## Debug log
[home-assistant_givenergy_local_2024-05-16T09-12-58.317Z.log](https://github.com/cdpuk/givenergy-local/files/15332563/home-assistant_givenergy_local_2024-05-16T09-12-58.317Z.log)

alex250472, 2024-05-21:
Upgraded from 1.9.1 to latest version and stopped working. Same hardware & firmware as above. Have rolled back to 1.9.1 and started working again.

Lmcde3, 2024-05-21:
Thanks for the update Alex. I'm also facing this big but prefer not to roll back. Am keeping an eye out for any further developments as I love this integration. Thanks all.

matteovescovi, 2024-06-26:
Hi,

I believe I may have found a workaround for the reported issue. I'm sharing the information here in case it helps you too.

I'm running with:

    Inverter model & generation: GIV-HY5.0
    Inverter firmware version: D0.450-A0.451
    Battery firmware version: 3015
    Home Assistant integration version: 2.0.4

I had (almost) the same issue. My logs showed the following warnings:

  2024-06-26 10:23:54.173 WARNING (MainThread) [custom_components.givenergy_local.coordinator] Failed to convert discharge_slot_2 from [0, 60]: minute must be in 0..59

This is similar to the warnings in the attached log:

  2024-05-16 00:42:47.327 WARNING (MainThread) [custom_components.givenergy_local.coordinator] Inverter model failed validation: Failed to convert discharge_slot_1 from [60, 2359]: minute must be in 0..59

and the givenergy_local integration would error out and give me ‘no devices or entities’ and ‘failed to setup, will retry’.

I used the givenergy cloud remote control on https://givenergy.cloud/ to read the Discharge 2 Settings.

In my case, the Discharge 2 End Time config was showing a value of --:--.

I set it to 00:00 and reloaded the integration.

After doing that, the integration is working correctly.

closed 2024-07-09 by cdpuk

cdpuk, 2024-07-09:
Checks have been added to look for that invalid '60' value in timeslot definitions. We'll assume this means "undefined" since the portal displays "--:--" in such cases. This should allow the integration to load with the timeslot sensors in the 'off' position until a valid time is set.

Thanks for the logs to help pinpoint this one.

PR #91 (2024-07-09): Handle invalid timeslot values
Fixes #82