cdpuk/givenergy-local #83: Unable to load integration with 2024.6 beta

chrisdwest1984, 2024-05-29:
## Hardware and versions

* Inverter model & generation: AC Coupled
* Inverter firmware version: D0.535-A0.535
* Battery firmware version: 3015
* Home Assistant integration version: 2.0.3

## Describe the bug

I've just installed the new HA  beta and now the integration fails to load due to an 'Invalid config' error. Log is below.

## Debug log

```text

Add your logs here.

Logger: homeassistant.util.package
Source: util/package.py:123
First occurred: 22:48:14 (3 occurrences)
Last logged: 22:48:31

Unable to install package pydantic==1.10.12: ERROR: Cannot install pydantic==1.10.12 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

tanathka, 2024-06-05:
Same error but with general release 2024.06 of HA
Inverter Model: GIV-HY5.0
Software Version: D0.450-A0.451
Battery Firmware Version: 3015
Home Assistant integration version: 2.0.3

timsweb, 2024-06-06:
Same here on the 2024.06 HA release. 

```Logger: homeassistant.setup
Source: setup.py:269
First occurred: 00:10:40 (1 occurrences)
Last logged: 00:10:40

Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic==1.10.12'].```

birdo78, 2024-06-06:
Same issue for me on 2024.6;

Core 2024.6.0
Supervisor 2024.06.0
Operating System 12.3
Frontend 20240605.0

`Logger: homeassistant.util.package
Source: util/package.py:123
First occurred: 06:48:17 (3 occurrences)
Last logged: 06:48:53

Unable to install package pydantic==1.10.12: ERROR: Cannot install pydantic==1.10.12 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts`

pskillen, 2024-06-06:
I've opened #85 to address this, but haven't been able to test that it won't break things for pre-2024.6 users. Don't suppose anyone has a test setup of HA that they haven't upgraded?

@cdpuk I've never ventured into HA plugin development. Is there a reason why everyone seems to pin their pip packages down to the build (`pydantic==1.10.12`)? Would `pydantic >=1.10,<1.11` be acceptable?

nimmat, 2024-06-06:
Same problem here with pydantic==1.10.12 with conflicting dependencies on 2024.6.0 
Had to revert back to 2024.5.* as the GiveEnergy Local plugin is very critical.

But thanks @cdpuk for this amazing integration. Hopefully this can be fixed quickly.

cdpuk, 2024-06-06:
> I've opened #85 to address this, but haven't been able to test that it won't break things for pre-2024.6 users. Don't suppose anyone has a test setup of HA that they haven't upgraded?
>
> @cdpuk I've [...]

... and to answer that. The reason for pinning the dependencies was original to stay closely aligned with the original `givenergy_modbus` project. Clearly this may no longer be the right thing to do, so perhaps needs revisiting. But thanks for sorting this one out 👍

chrisdwest1984, 2024-06-06:
Thanks for sorting!

fbutler, 2024-06-06:
Thanks from me as well.

LukeAllinson, 2024-06-07:
Thanks so much for the speedy fix 👍

capsel22, 2024-06-10:
Hi. Im on 2024.5.1 and getting the "Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic==1.10.15']." error when updating to 2.0.4.

I can see this issue is marked as fixed but it's not working for me, any ideas?

cdpuk, 2024-06-10:
> Hi. Im on 2024.5.1 and getting the "Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic==1.10.15']." error when updating to 2.0.4.
> 
> I can see this issue is marked as fixed but it's not working for me, any ideas?

Are you on a container based install? There seems to be some combination of HA/HACS installs that don't handle dependency upgrades well. The release notes for 2.0.4 contain a couple of suggestions. Please feed back if these or any other approach works.

capsel22, 2024-06-10:
> Are you on a container based install? There seems to be some combination of HA/HACS installs that don't handle dependency upgrades well. The release notes for 2.0.4 contain a couple of suggestions. [...]

thanks for fast reply.
I'm running HA OS with HACS. 

I managed to SSH into the HA and ran "pip list " I get the below 

~ # pip list
Package    Version
---------- -------
awake      1.0
packaging  24.0
pathspec   0.12.1
pip        24.0
pulsemixer 1.5.1
pyparsing  3.1.2
PyYAML     6.0.1
setuptools 69.5.1
yamllint   1.35.1

Im not HA or ssh wizard, so I'm a bit lost here. Would updating to 2023.6 resolve this?

capsel22, 2024-06-10:
looks like my HA OS 2023.5 didnt have pydantic at all which failed to start 2.0.4.
When I updated to 2023.6 it worked, didnt need to do anyhing else.

mdglawson, 2024-06-10:
> looks like my HA OS 2023.5 didnt have pydantic at all which failed to start 2.0.4. When I updated to 2023.6 it worked, didnt need to do anyhing else.

I was the same - running 2023.5.5 and getting this pydantic error, upgrading to 2023.6.1 and installing v2.0.4 worked for me