cdpuk/givenergy-local #87: Update to HA 2024.07 Creates failure to load due to updated pedantic package

bluesheep, 2024-07-03:
## Hardware and versions
HA 2024.07.0 
Proxmox VM

* Inverter model & generation: N/A
* Inverter firmware version:
* Battery firmware version:
* Home Assistant integration version:

## Describe the bug

Custom component fails to start due to error loading.

```
Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic==1.10.15']
```

## Debug log

N/A

bluesheep, 2024-07-03:
See [This chore](https://github.com/cdpuk/givenergy-local/commit/7c35d8784f4a6ee39ff2315e649deb8c70c8473a) (Sorry, I'm new to Github)

bluesheep, 2024-07-03:
I have submitted a PR that I think addresses it, at least tactically.

russdan, 2024-07-04:
Needs to bump to `1.10.17` now

What's the long term fix needed to avoid this? Just bumping pydantic versions in `manifest.json` and `requirements.txt` can't be the strategic fix?  I'm happy to have a stab at a proper fix, but not sure what needs doing!  Or do we just remove specific versions from these files, essentially getting 'latest'?

ianfretwell, 2024-07-04:
Can someone just advise where I actually find the _requirements.txt_ file in order to edit it - searching my homeassistant persistent folder comes up blank - whereas the custom_components is obvious.

Thanks.

poolski, 2024-07-05:
> Can someone just advise where I actually find the requirements.txt file in order to edit it - searching my homeassistant persistent folder comes up blank - whereas the custom_components is obvious.

The `requirements.txt` is a module dependency: it's for the project maintainers to update - `manifest.json` is the only file you need to worry about.

poolski, 2024-07-05:
@cdpuk I'm not sure if this is related, but I'm getting issues where the integration fetches data from the inverter at boot time and publishes it to MQTT **once** and then it's _crickets_. 

I'm not sure yet if it's an MQTT problem (I'm using EMQX rather than Mosquitto in HA), but none of my MQTT settings have changed, so I'm finding it a bit odd. 

Happy to provide more info if I can get some idea of where to look/what to investigate. 🤔

closed 2024-07-09 by cdpuk

cdpuk, 2024-07-09:
Strategic fix now merged in #90:

- Removed pin to a specific `pydantic` version. HA core pins this in [package_constraints.txt](https://github.com/home-assistant/core/blob/dev/homeassistant/package_constraints.txt), so we can't fight it. The new approach should work from 2023.10 onwards, or at least until the HA pin moves to `pydantic` 2 onwards.
- Removed other dependencies that were only included in commented out debug code within `givenergy_modbus`.

russdan, 2024-07-10:
Awesome, thanks @cdpuk 👍   Looking forward to the next HA release to test it out :)

PR #90 (2024-07-09): Rethink pinned dependencies to avoid conflicts
Fixes #87