cdpuk/givenergy-local #89: Error Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble

jordmccord, 2024-07-04:
I can't get the integration past the wizard loading spinner and keep getting the following error:

`Error Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble`

HA Version:
```
Core: 2024.7.0
Supervisor: 2024.06.2
Operating System: 12.4
Frontend: 20240703.0
```

nimmat, 2024-07-04:
Same here: Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble
Core: 2024.7.0
Supervisor: 2024.06.2
Operating System: 12.4
Frontend: 20240703.0

Happened after I upgraded to 2024.7.0

dazrave, 2024-07-05:
I assume this is the issue I've got. Just upgraded and also only just installed this in HACS but can not get it working at all. Same error as above.

cdpuk, 2024-07-09:
Believed to be a duplicate of #87

closed 2024-07-09 by cdpuk