cdpuk/givenergy-local #93: Charge and Discharge Slots become unavailable for a minute or so

MarkTimStafford, 2024-09-05:
Inverter Model  GIV-HY3.6
Software Version  D0.449-A0.449
320 Ah | 16.38 kWh | (2 Packs)

Inverter Firmware Version  D0.449-A0.449
Battery Firmware Version   3017
---

GivEnergy-Local - version 2.0.6

My Charge and Discharge slots become unavailable for a minute or so and then come back with the same value.
HA Entity view attached
[image]

I have a dedicated IP address and pinging the IP seems fine and fast
Pinging 192.168.1.101 with 32 bytes of data:
Reply from 192.168.1.101: bytes=32 time=8ms TTL=64
Reply from 192.168.1.101: bytes=32 time=2ms TTL=64
Reply from 192.168.1.101: bytes=32 time=3ms TTL=64
Reply from 192.168.1.101: bytes=32 time=5ms TTL=64

Nothing related in HA Logs

Any ideas???

cdpuk, 2024-09-09:
This is normally due to transient issues talking to the inverter, and is fairly common presumably due firmware issues. Can you attach any debug logs while such events happen?

MarkTimStafford, 2024-09-10:
I cannot provide the debug logs as I have stopped relying on the Battery Charge Slots? I think that for this case to proceed I would need more information and diagnostic material, which I cannot easily provide. So I have closed the issue.