cdpuk/givenergy-local #94: Unable to activate timed charging

kylegordon, 2024-09-09:
Hi, 

Trying to take advantage of Octopus free electricity time slots, but I can't seem to get the integration to apply the changes to the inverter.

Hybrid Gen 3 5.0kW, Firmware: D0.309-A0.309 
2x Giv-Bat 9.5, Firmware: 3017

Givenergy-local v2.1.0-beta4

Triggering the following action results in no change to the inverter in the mobile app or in the GE portal, even after a few minutes:

```action: givenergy_local.enable_timed_charge
data:
  device_id: 636335f26ef06930329cf34272a13502
  start_time: "13:00:00"
  end_time: "14:00:00"
```

I have attached the debug logs, from enabling debug, reloading integration, to submitting the action.

I think the relevant lines may be

```
2024-09-09 22:43:01.405 DEBUG (MainThread) [custom_components.givenergy_local.coordinator] Finished fetching Inverter data in 0.922 seconds (success: True)
2024-09-09 22:43:01.420 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_1 at 2024-09-10 13:00:00
2024-09-09 22:43:01.462 DEBUG (MainThread) [custom_components.givenergy_local] Scheduled next update for charge_slot_2 at 2024-09-10 00:01:00
```
[home-assistant_givenergy_local_2024-09-09T21-46-12.956Z.zip](https://github.com/user-attachments/files/16936955/home-assistant_givenergy_local_2024-09-09T21-46-12.956Z.zip)

The second battery schedule does not seem correct.

Happy to experiment!

kylegordon, 2024-09-15:
Heya,

Sorry about the wait in getting back to you.

I've downgraded to v2.0.6, and am now also testing with just a plain/brand new home-assistant config in docker. Unfortunately it appears to be the same. I can't get new schedules to stick or enable, and I can't get existing ones to be disabled either, using the Disable Timed Charging call. 

Maybe I'm doing something wrong and have misread the docs? I don't use any features other than reading entities at this point.

[home-assistant_givenergy_local_2024-09-15T19-50-01.044Z.zip](https://github.com/user-attachments/files/17007205/home-assistant_givenergy_local_2024-09-15T19-50-01.044Z.zip)

I can edit the charging times in the GE web UI and the HA logs (the "Scheduled next update" lines) reflect it correctly. Triggering an action with the above YAML also has the logs showing the requested times, but it doesn't appear in the GE mobile app or web UI.

kylegordon, 2024-09-22:
The confusion continues, I've just spotted in my monitoring that the battery is clearly charging between 1300 and 1400, yet the GE website and app both state that charging is disabled, and if it were it would be between 1800 and 2000.

I think it might be time for a restart or something...

kylegordon, 2024-10-20:
I just upgraded Home-Assistant from something like 2024.09 to the latest 2024.10 and thought I'd try the timed charge again... and it worked. 

Will try disabling it again after today's free energy slot and if I remember will do a few more tests this week. Can hopefully close it soon and chalk it up to GE weirdness!

cdpuk, 2024-11-07:
Unfortunately it's hard to comment on how the app and web portal work - they won't be expecting a third party to be updating values like the integration does, so if/when the updates appear in the app/web UI is a bit of an unknown.

If the inverter is doing what the integration tells it, that's the main thing. I'll close this for now, but feel free to reopen if there's anything that the integration could do to improve the situation.

closed 2024-11-07 by cdpuk