cdpuk/givenergy-local #96: Simplified 'Stop/Start Charging ability ?

ianfretwell, 2024-09-17:
Is there anyway to have a simple as it comes 'start charging', 'stop charging' ?    (i.e. the same options shown in the app when you press on the green 'play' button at the bottom/centre of the app).

My use case is because the AC3 doesn't have two charge time slots - only the one.   What I'd like to be able to do is charge from 00:00 to 02:00 and again from 06:00 to 07:00    (this is trying to get the most from a cheap overnight rate whilst only having a 5.2 battery).  If I could do this by essentially just turning an entity (switch) on and off that would be easy to do.

Or should I be doing this some other way? - although if so I don't see how I am supposed to update the charge slot programatically - if I could I guess I could just set it to 00:00-02:00 at 23:00 and then set it to 06:00-07:00 at 05:00.   Am I missing the obvious ?

Thanks

ianfretwell, 2024-09-17:
Or is this an alternative sequence (and if so how do I actually do it?): -

Single timed charge setting 00:00 - 07:00
Set charge power to zero at 02:00
Set charge power to maximum at 06:00

Then leave charge power at maximum to get whatever I can from solar during the day ? Again - sounds simple but I don't know how to do it.

pskillen, 2024-09-17:
If it helps, I have charging slot 1 set to 00:00-23:59, and I just use HA to toggle the AC Charging Enabled switch. This essentially does away with the scheduling logic inside the inverter and replaced it with my HA automations, which works great for me.

Note the same isn't true for DC Discharging! If you're 'inside' a discharging slot, but the DC Discharging switch is off, the inverter no longer supplies the house, which usually isn't what you want

ianfretwell, 2024-09-17:
Ah, right - so the AC Charging Enabled switch becomes a   is-grid-charging/is-not-grid-charging    switch?

And when it's 'not charging' everything works as per normal eco-mode would ?

pskillen, 2024-09-17:
For AC Charging, yes, if you turn off the switch then it just goes back to eco mode (but not the same for DC Discharging, which is very annoying!)

ianfretwell, 2024-09-20:
@pskillen Many thanks for your suggestion - it's worked a treat!

cdpuk, 2024-11-09:
Sounds like this has been solved.

Unfortunately every inverter is a bit different, so trying to build high level functions on top of what we have can be quite difficult, especially when testing is impossible without owning the relevant bit of kit. In this case, I suspect the "pause" you see in the app refers to battery pause mode, which is only supported on more recent equipment. The integration aligns more closely with the "Remote Control" view in the web portal, which reads & writes lower level values to control the inverter. Hopefully this approach continues to work for most users and HA give us enough power through automations to achieve what's needed.

closed 2024-11-09 by cdpuk