cdpuk/givenergy-local #97: Charging not stopping when going into Eco mode

hilburn, 2024-10-13:
## Hardware and versions

* Inverter model & generation:
* Inverter firmware version:
* Battery firmware version:
* Home Assistant integration version:

## Describe the bug

I've been using Eco Mode to exit out of Timed Charging Mode for a while now, but recently due to a change in my control scheme I've changed the Timed Charging to target 100% and cancel out of it early with a boolean triggering the automation (state>true: turn on charging, state>false: turn on eco mode)

However last night I noticed that the charging went beyond the return to Eco mode (green is the window where the boolean is true)

[image]

This seems to be because the Activate Eco Mode doesn't reset AC charging to off, or disables the Battery Charge Slot 1 - so while it's still in the charge slot (2-5am) I set, it still charges the battery despite nominally being in Eco mode

I'm not sure what's more sensible at this point, either changing the controls.md to reflect the additional steps to actually turn off Timed Charging Mode, or to change the Eco mode to turn AC charging and battery charge slot 1 off

## Debug log

```

This isn't the sort of issue that requires debug logs

```

cdpuk, 2024-11-10:
The behaviour of the "Activate Eco Mode" service is by design to match the behaviour of the GivEnergy app/portal. It's completely valid to have Eco mode on or off at the same time as AC charging being on or off; it's only when you're within the AC charging timeslot does charging actually happen.

There does appear to be a copy & paste mistake in `controls.md`, which incorrectly states that AC charging can be turned off by activating Eco mode. I'll make an update to fix that. The only way to turn it off is via `switch.battery_ac_charging`.

closed 2024-11-10 by cdpuk

hilburn, 2024-11-10:
That's fair! I wasn't sure if it was intended behaviour given it went against the docs so fixing the docs is perfectly valid :)