cdpuk/givenergy-local #98: Setup in HA fails, despite correct IP address

WinniTheis, 2024-10-27:
## Hardware and versions

* Inverter model & generation: GIV-3HY-8
* Inverter firmware version: DA0.004-DD0.004-A0.004
* Battery firmware version: 4
* Home Assistant integration version: 2.0.6

## Describe the bug

I cannot add the integration, because when I enter the IP address, it just times out. I checked that I can actually connect to the inverters UI in a browser, and that works fine, it just takes quite a while to get the login. Any idea what might be going wrong? I looked in several logs in HA, but none of them seems to capture errors in installation of integrations. Is there maybe a wait time in the install script, that could be increased? 
Thanks,
Winfried

## Debug log

```text

Could not find any log containing messages.

```

cdpuk, 2024-10-30:
You may be the first person to attempt to use a 3 phase inverter with the integration, so this is a bit of an unknown. There will certainly be some logs - either containing the timeout error or showing a successful connection. One possible outcome is a successful connection, but then something going wrong if the protocol is different - such detail may not appear in the UI.

You could also try the AIO version, available as a beta release via HACS. This was originally created to add AIO support, but there's a chance that your inverter communicates in the same way.

Ryuisnod, 2024-11-05:
Ive just had a 3-phase inverter installed and have the same issue, if I can share anything I will

WinniTheis, 2024-11-06:
Thank you for the feedback! Any hint where to find the logs would be appreciated. I will have a look at the beta version of AIO. Cheers,Winfried ________________________________"All models are wrong, but some are useful!" George Box

cdpuk, 2024-11-09:
The new v2.1.0 release has changed quite a lot, so please do try again with that and post back here.

crassay, 2025-07-09:
I just tried with the latest release and inverter version GIV-3HY-15. The settup of the intergration did not work. The following is logged in Home Assistant:

```
This error originated from a custom integration.

Logger: custom_components.givenergy_local
Source: custom_components/givenergy_local/config_flow.py:47
integration: GivEnergy Local (documentation, issues)
First occurred: 21:41:31 (2 occurrences)
Last logged: 21:41:47

Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 47, in async_step_user
    serial_no = await read_inverter_serial(user_input)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/config_flow.py", line 23, in read_inverter_serial
    await client.detect_plant()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 106, in detect_plant
    await self.refresh_plant(True, timeout=timeout, retries=retries)
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 190, in refresh_plant
    await self.execute(reqs, timeout=timeout, retries=retries)
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 373, in send_request_and_await_response
    raise asyncio.TimeoutError()
TimeoutError
``` 

And
```
This error originated from a custom integration.

Logger: custom_components.givenergy_local.givenergy_modbus.client.client
Source: custom_components/givenergy_local/givenergy_modbus/client/client.py:367
integration: GivEnergy Local (documentation, issues)
First occurred: 21:41:31 (14 occurrences)
Last logged: 21:41:48

Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x33 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60) after 4 tries at 1s, giving up
```

mg-nz, 2026-03-15:
Has anyone made any progress on this? I'm having the same issue with a 3 phase inverter (GIV-3HY-20). 

This is using version 2.2.2:

```
This error originated from a custom integration.

Logger: custom_components.givenergy_local
Source: custom_components/givenergy_local/config_flow.py:47
integration: givenergy_local (documentation, issues)
First occurred: March 9, 2026 at 9:34:42 PM (2 occurrences)
Last logged: 10:35:42 AM

Failed to validate inverter configuration
Traceback (most recent call last):
  File "/config/custom_components/givenergy_local/config_flow.py", line 47, in async_step_user
    serial_no = await read_inverter_serial(user_input)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/givenergy_local/config_flow.py", line 23, in read_inverter_serial
    await client.detect_plant()
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 106, in detect_plant
    await self.refresh_plant(True, timeout=timeout, retries=retries)
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 190, in refresh_plant
    await self.execute(reqs, timeout=timeout, retries=retries)
  File "/config/custom_components/givenergy_local/givenergy_modbus/client/client.py", line 373, in send_request_and_await_response
    raise asyncio.TimeoutError()
TimeoutError
```

cdpuk, 2026-08-19:
Closing as stale

closed 2026-08-19 by cdpuk