cdpuk/givenergy-local #105: Fails after initial connection

jon798, 2024-11-21:
Hello,

* Inverter model & generation: GIV-HY3.6. (Gen 3 hybrid 3.6 kW)
* Inverter firmware version: D0.309-A0.309
* Battery firmware version: (missing from portal/app)
* Home Assistant integration version: 2.0.1

## Describe the bug

When the integration starts, devices and sensors are created for the inverter and the 3x 5.12kW Gen 3 batteries that are installed. Shortly afterwards (e.g. 10 seconds), the sensors all report 'unavailable' and don't recover.  Reloading the integration repopulates the sensros for a few more seconds until they again all become 'unavailable'.

## Debug log

[home-assistant_givenergy_local_2024-11-21T08-49-07.360Z.log](https://github.com/user-attachments/files/17842795/home-assistant_givenergy_local_2024-11-21T08-49-07.360Z.log)

If there's anything else I can do to help diagnose/troubleshoot, please let me know.

Thanks!

jon798, 2024-11-24:
To follow up on this - everything appears to have sprung into place after GivEnergy pushed through a battery firmware update from v4004 to v4007. So probably can close this issue.

closed 2024-11-24 by jon798

cdpuk, 2024-11-25:
Good to know. Thanks for the update.