cdpuk/givenergy-local #108: Dependency issue with 2025.1

alphasixtyfive, 2024-12-30:
It looms like ther eis a dependency issue with `pydantic` in 2025.1.X.

```
Logger: homeassistant.setup
Source: setup.py:269
First occurred: 10:44:26 (1 occurrences)
Last logged: 10:44:26

Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic>=1.10.12,<2'].
```

```
Logger: homeassistant.util.package
Source: util/package.py:152
First occurred: 10:44:25 (3 occurrences)
Last logged: 10:44:26

Unable to install package pydantic>=1.10.12,<2: × No solution found when resolving dependencies: ╰─▶ Because you require pydantic>=1.10.12,<2 and pydantic==2.10.4, we can conclude that your requirements are unsatisfiable.

```

chrisdwest1984, 2024-12-30:
Same issue here. Home Assistant has moved to pydantic V2. I tried modifying the manifest.json file but it leads to different errors.

oliwarner, 2025-01-03:
Libraries that do this can be a bit of a pain but [Pydantic has a `pydantic.v1` import where you can import most of the original features](https://docs.pydantic.dev/latest/migration/#code-transformation-tool). This can make it easier to use the v2+ with old code.

oliwarner, 2025-01-03:
Can confirm my fork _Works For Me_™ on 2025.1, phew.

I've no plan to maintain it further than I need it (I have a PR in) but anyone who doesn't want to downgrade HA can install it by adding it as a custom repo in HACS: `https://github.com/oliwarner/givenergy-local` (integration). Once this is merged, or otherwise improved, I'll nuke my repo and HACS should revert.

ianfretwell, 2025-01-04:
I've put your changes from your PR into the files manually - works a treat - many thanks.

russdan, 2025-01-04:
> I've put your changes from your PR into the files manually - works a treat - many thanks.

+1 ..... integration works again with the #111 PR changes, thanks @oliwarner

capsel22, 2025-01-04:
> I've put your changes from your PR into the files manually - works a treat - many thanks.

would you mind explaining how?

ianfretwell, 2025-01-04:
Go to the PR here[https://github.com/cdpuk/givenergy-local/pull/111/commits/2ee475f9d7e4ff45cce06c08de93780d9b715ad7](url)

That shows you each of the files you'll need to edit complete with the from (-) /to (+) change - which in every case is just adding '.v1' after the word pydantic.

You can ignore the requirements.txt file.

russdan, 2025-01-04:
> > I've put your changes from your PR into the files manually - works a treat - many thanks.
> 
> would you mind explaining how?

I would say if you don't know how then it's risky..... as @ianfretwell has said, look at PR #111 and see the changes in `Files Changed` (there are 6 files changed, but you can ignore `requirements.txt`),. and then use something like the Terminal to edit (`nano` is available from the Terminal) each file and make the changes (if you see each change has red `-` which are the lines deleted and green `+` for the lines added.  Once you've updated all 5 files then restart HA and it should start the integration.

However, if this is new and you're not familiar with this type of process then I would say wait for the PR to get approved so you pick up the official version, or you risk breaking things if the changes aren't applied correctly (ie you could possibly stop your whole HA instance, maybe, possibly!)..... its probably the case you'll only break the givenergy-local integration which of course is already broken...

capsel22, 2025-01-04:
Thank you both, I worked it out. Made a backup of original files, no bother.

closed 2025-01-04 by cdpuk

richatnort, 2025-01-04:
Is there a ETA when a new version will get pushed to HACS? I can't see a 2.2.1 or a BETA there.

cdpuk, 2025-01-04:
This is now on HACS as v2.2.1 as of 5 minutes ago 😄

PR #111 (2025-01-04): Compat with Pydantic v2 for ha2025
This is a quick code patch to use the shims in pydantic 2+ that's being used in homeassistant 2025.1

Fixes #108 

I've added `try/except ImportError` shims to keep it backwards compatible with 2024.