cdpuk/givenergy-local #109: Errors since midnight on 31/12

hilburn, 2024-12-31:
## Hardware and versions

* Inverter model & generation:  GIV HY5 GEN3
* Inverter firmware version: D0.309-A0.309
* Battery firmware version: 3017
* Home Assistant integration version: 2.2.0

## Describe the bug

This feels like a GivEnergy error but it appears to be reporting the date as 1/13/25 rather than 31/12/24 but I'm not 100% sure and thought you should know about it - all data from the inverter has stopped updating in ha, though the app is correct, and services are still working properly

## Debug log

```text

This error originated from a custom integration.

Logger: custom_components.givenergy_local.coordinator
Source: custom_components/givenergy_local/coordinator.py:162
integration: GivEnergy Local (documentation, issues)
First occurred: 00:04:28 (3246 occurrences)
Last logged: 06:46:17

Failed to convert system_time from [25, 13, 1, 6, 45, 40]: month must be in 1..12
Failed to convert system_time from [25, 13, 1, 6, 45, 44]: month must be in 1..12
Failed to convert system_time from [25, 13, 1, 6, 45, 58]: month must be in 1..12
Failed to convert system_time from [25, 13, 1, 6, 46, 2]: month must be in 1..12
Failed to convert system_time from [25, 13, 1, 6, 46, 6]: month must be in 1..12

```

Full debug log is generating currently, will update when it's done

jasonwragg, 2024-12-31:
If you want a quick fix i did this to get it working:

edit line 137 of /givenergy_local/givenergy_modbus/model/register.py

```
 @staticmethod
  def datetime(year, month, day, hour, min, sec) -> Optional[datetime]:
      """Compose a datetime from 6 registers."""
      if None not in [year, month, day, hour, min, sec]:
          return datetime(year + 2000, month, day, hour, min, sec)
      return None
```

change it to this:

```
@staticmethod
def datetime(year, month, day, hour, min, sec) -> Optional[datetime]:
    """Compose a datetime from 6 registers."""
    if None not in [year, month, day, hour, min, sec]:
      if month ==13
        month = 12
      return datetime(year + 2000, month, day, hour, min, sec)
    return None
```

Its worked for me.

andynash, 2024-12-31:
The cause appears to be the same as this GivTCP issue, I can report the suggested fix (update inverter time from the portal) worked for me:

https://github.com/britkat1980/giv_tcp/issues/318

In the portal, go to your inverter, click the cog for settings and send the time/date update, then reload the integration.

Weird how GivEnergy struggle so much with the date at New Year. Is not setting the month to 13 too much to ask, especially the day *before* the year actually ends 😂

hilburn, 2024-12-31:
Good find!
sending the date/time update worked a treat and I reported it to GE
I find it especially hilarious as it went 12/24 - > 13/25 which is just not how dates work at all!

richatnort, 2024-12-31:
Thanks for the instructions Andy, that worked for me! Wasn’t sure what portal meant but it’s just the giv energy cloud site.

alex250472, 2024-12-31:
Thanks Andy, fix worked for me.

cdpuk, 2025-01-04:
That's quite an impressive bug. I'm not going to do anything about it this year, as clearly the correct approach here is a firmware fix. If it's still an issue next year, then let's reconsider. For now I've updated the troubleshooting docs.

closed 2025-01-04 by cdpuk

gcoan, 2025-01-04:
> That's quite an impressive bug. I'm not going to do anything about it this year, as clearly the correct approach here is a firmware fix. If it's still an issue next year, then let's reconsider. For now I've updated the troubleshooting docs.

From the discussion on the Givenergy community forum it rather looks like it is a leap year bug that the inverter firmware doesn't allow for the extra leap day when rolling over to the start of the new year, i.e. it does it one day early and rolls over to the 13th month !

whether this gets fixed in all givenergy firmware versions I personally remain sceptical, especially the older AC3 and Gen 1 hybrids, but who knows

hilburn, 2025-01-05:
I think the fix is pretty simple
If an error is thrown decoding the date, use `now()` instead