cdpuk/givenergy-local #112: Givenergy-local not uploading after homeassistant update

compo-23, 2025-01-04:
## Hardware and versions

* Inverter model & generation:
* Inverter firmware version:
* Battery firmware version:
*
Core2025.1.0
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250103.0

## Describe the bug

Failure to start after homeassistant upgrade, presumably because pydantic version has changed to 2 which is apparently not backwards compatible

## Debug log

```text

Add your logs here.
Logger: homeassistant.setup
Source: setup.py:269
First occurred: 15:12:37 (1 occurrences)
Last logged: 15:12:37

Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic>=1.10.12,<2'].
```

oliwarner, 2025-01-04:
This is a duplicate of #108 - which has a fix PR #111 that you can apply by temporarily using my repo, or manually applying the patches until this is fixed.

cdpuk, 2025-01-04:
Fixed and released under v2.2.1

cdpuk, 2025-01-04:
Duplicates #108

closed 2025-01-04 by cdpuk