cdpuk/givenergy-local #115: Intermittent Entities - modbus timeout

jragarw, 2025-01-05:
hi all - been on 1.9 for ages - due to the entities intermittently going unavailable - however the new HA version necessitated an update to v2.2+ - and now the entity intermittent issue has returned. I have attempted to move the polling/update to 30s but to no avail. 

I've provided a debug log below

Inverter Model
GIV-HY5.0
Inverter Firmware Version
D0.450-A0.451
Last Update Time
2025-01-05 00:19:14
Battery Firmware Version
3017

debug log:

2025-01-05 02:37:04.763 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:37:04.766 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:37:05.014 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:37:42.032 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:39:44.762 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:39:44.763 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:39:44.765 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:39:45.013 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:40:22.046 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:42:54.762 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:42:54.762 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:42:54.764 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:42:55.013 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:43:32.038 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:46:04.761 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:46:04.762 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:46:04.765 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:46:05.013 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:46:42.038 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:48:44.765 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:48:44.766 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:48:44.769 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:48:45.015 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:49:22.041 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:51:54.761 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:51:54.762 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:51:54.763 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:51:55.013 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:52:32.038 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:55:04.762 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:55:04.762 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:55:04.765 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:55:05.014 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:55:42.030 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 02:57:44.761 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up
2025-01-05 02:57:44.762 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Closing connection due to expected error: 
2025-01-05 02:57:44.763 ERROR (MainThread) [custom_components.givenergy_local.coordinator] Error fetching Inverter data: Connection closed due to expected error
2025-01-05 02:57:45.014 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
2025-01-05 02:58:22.040 WARNING (MainThread) [custom_components.givenergy_local.givenergy_modbus.client.client] Timeout awaiting 2:3/ReadHoldingRegistersResponse(slave_address=0x32 base_register=300) after 4 tries at 1s, giving up
2025-01-05 03:00:02.355 WARNING (MainThread) [homeassistant.components.samsungtv] Failed to get remote for 192.168.1.151: ConnectionClosedError(None, Close(code=<CloseCode.PROTOCOL_ERROR: 1002>, reason='invalid status code'), None)
[... cut ...] T03:06:12.474889+00:00. Please create a bug report at https://github.com/cdpuk/givenergy-local/issues
2025-01-05 03:10:10.224 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.consumption_total from integration givenergy_local has state class total_increasing, but its state is not strictly increasing. Triggered by state 4777.5 (4777.6) with last_updated set to 2025-01-05T03:06:12.488084+00:00. Please create a bug report at https://github.com/cdpuk/givenergy-local/issues

alex250472, 2025-01-06:
I am having the same problem. 1.91 was working.

Inverter Model
GIV-AC3.0

Grease-Monkey80, 2025-01-12:
Having the same issue, and as above it’s causing havoc with automations.

cdpuk, 2025-01-17:
Hi @jragarw, you appear to have exactly the same inverter and firmware versions as me, which is probably a first on here. However, as you can probably guess, mine is rock solid on the latest version.

Something definitely isn't right in those logs, as it appears to be looping on `Closing connection due to expected error`. This should be triggering a reconnect from scratch.

Are you able to capture some debug logs as the integration starts up? I could do with seeing these two lines onwards:
```
Connection established to <ip>:8899
Performing model detection
```

Reloading the integration should trigger this.

jragarw, 2025-01-17:
@cdpuk Sure thing! hopefully I have done this correctly - let me know if not. I reloaded the integration and left it logging until the entities became unavailable.

[2025-01-17 211201.214 INFO.txt](https://github.com/user-attachments/files/18459849/2025-01-17.211201.214.INFO.txt)

ianfretwell, 2025-03-07:
Just wondering if I'm now seeing this same issue

AC3                  D0.209-A0.209
GivBat 5.2        3020

Since upgrading to 3020 two days ago I seem to be getting a lot of these in the basic logs  (and the entities keep disappearing for short amounts of time):-

This error originated from a custom integration.

Logger: custom_components.givenergy_local.givenergy_modbus.client.client
Source: custom_components/givenergy_local/givenergy_modbus/client/client.py:367
integration: GivEnergy Local (documentation, issues)
First occurred: 5 March 2025 at 21:43:21 (270 occurrences)
Last logged: 09:53:01

Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x34 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x35 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60) after 4 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60) after 3 tries at 1s, giving up
Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0) after 3 tries at 1s, giving up

jragarw, 2025-04-21:
[image]

has anyone found a workaround for the issues on this? It seems like it is getting worse for me - specifically the battery percentage entity.

pskillen, 2025-04-21:
It's not a good workaround @jragarw, but I've created a template sensor that looks at the battery percent. If it's defined, I use the current value, if it's not defined I use the previous template value

```jinja2
{% set state = states('sensor.battery_percent') %}

{% if state | is_number and (state | float) > 0 %}
{{ state | float }}
{% else %}
{{ states('sensor.givbatt_percent_available') | float }}
{% endif %}
```

Grease-Monkey80, 2025-04-21:
I also found a non-perfect work around by taking short term (30sec) averages of the entities I use in automations to give me steady values.

alex250472, 2025-05-07:
I have been using the GE Cloud integration. Not ideal as needs the internet, but has been working fine for now.

https://github.com/springfall2008/ge_cloud

Peter-Maguire, 2025-06-08:
I'm also getting these intermittent timeouts after updating my inverter to D0.316-A0.316 - seems to stay connected for around 5 minutes max

alex-taylor, 2025-08-06:
The failures happen at approximately 5-minute intervals.  Requests made in between them succeed.

This sounds like it could be a rate-limiting thing.

alex-taylor, 2025-08-06:
Ah, good, I was just looking at the code to see how to do that! :-)

From the logs, it appears that it polls the inverter every 10 seconds and data comes back pretty much instantly when it works.

alex-taylor, 2025-08-06:
Hmm, just found this in a givenergy-modbus issue: https://github.com/dewet22/givenergy-modbus/issues/19#issuecomment-1401099317

alex-taylor, 2025-08-06:
Righty :)  After some tinkering, I have a potential fix for this.

Taking a cue from the GivTCP code, I've added explicit handling for TimeoutError so that it now closes and reopens the client connection to try again.  If this happens too many (>3) times then we should still fail with an error as before; in the meantime, this has eliminated the 'unavailable' sensor problem for me.

The code is in my fork here, if anyone would like to try it out: https://github.com/alex-taylor/givenergy-local

Peter-Maguire, 2025-08-06:
Awesome, I'll try your fork. Thank you

fbutler, 2025-08-06:
It looks like the changes in the fork may have resolved this. My graphs are now showing solid lines since I configured  it.

intrinseca, 2025-08-07:
[image]

Working for me too, you can see exactly when I installed the update! Thank you so much @alex-taylor for finding a fix/workaround!

jragarw, 2025-08-22:
Hi! Stupid question time :) do you know how to change forls in HACS/home assistant? I've been googling/trying a bunch of stuff but to no avail using a whole bunch of different values to try and force the update using update.install - apologies but does anyone know where I'm going wrong here?  thanks!!

[image]****