cdpuk/givenergy-local #116: CRC 0x0000 does not match calculated CRC

capsel22, 2025-01-07:
## Hardware and versions

* Inverter model & generation: AC Gen 1 3.0kW
* Inverter firmware version: D0.535-A0.535
* Battery firmware version: 3015
* Home Assistant integration version: 2025.1

## Describe the bug

For a few days I keep getting those CRC warning in system log. Not sure what they are about, integration seems to work as far as I can tell.

```text

This error originated from a custom integration.

Logger: custom_components.givenergy_local.givenergy_modbus.client.client
Source: custom_components/givenergy_local/givenergy_modbus/client/client.py:380
integration: GivEnergy Local (documentation, issues)
First occurred: 6 January 2025 at 21:40:29 (4 occurrences)
Last logged: 05:33:12

Decoding response (possibly 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=60)) resulted in exception: supplied CRC 0x0000 does not match calculated CRC 0x2e5a
Decoding response (possibly 2:4/ReadInputRegistersResponse(slave_address=0x32 base_register=0)) resulted in exception: supplied CRC 0x0000 does not match calculated CRC 0xe274
Decoding response (possibly 2:4/ReadInputRegistersResponse(ERROR slave_address=0x32 base_register=0 register_count=0)) resulted in exception: supplied CRC 0xa681 does not match calculated CRC 0x26e1
Decoding response (possibly 2:4/ReadInputRegistersResponse(ERROR slave_address=0x32 base_register=236)) resulted in exception: supplied CRC 0x6765 does not match calculated CRC 0xe705

```

capsel22, 2025-01-15:
they just keep appearing, not really sure what this means
[image]

cdpuk, 2025-01-17:
Hi @capsel22. CRC checking is a relatively new feature, and helps prevent invalid data propagating through to HA. The docs contain a short section on this, which could probably do with expanding:
https://github.com/cdpuk/givenergy-local/blob/master/docs/troubleshooting.md#crc-errors

Unfortunately the modbus connections seem prone to occasional errors, so these are unavoidable. Without these checks you'd instead see impossible values appearing, e.g. battery temperature = 9346 degrees. Previously, the integration used to try and detect these by defining likely ranges of values, but these are no longer needed thanks to the CRC checks.

If you're only getting a few per day, that's pretty normal. That's a tiny fraction of the many thousands of messages being checked. If you completely stop receiving data due to failed CRC checks, then we definitely have a problem.

closed 2025-01-17 by cdpuk