cdpuk/givenergy-local #117: entity "not loaded".  Unable to delete, unable to reinstall

garethedwards99, 2025-01-27:
## Hardware and versions

Home assistant
Core
2025.1.4
Supervisor
2024.12.3
Operating System
14.1
Frontend
20250109.2

HACS 
2.0.4

Inverter Model
GIV-HY5.0
Software Version
D0.450-A0.451

Inverter Firmware Version
D0.450-A0.451
Last Update Time
2024-10-30 18:25:34
Battery Firmware Version
3017

## Describe the bug

Since upgrade HA the givenergy entity will not load "not loaded"
I am unable to delete the givenergy integration
I am unable to reinstall the givenergy integration (Error Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble)

I have update HA and HACS to the latest version
I have checked that RA is enabled in the givenery portal.

You givenergy integration has been excellent and I would like t resolve the issue.

## Debug log

Logger: homeassistant.setup
Source: setup.py:269
First occurred: 10:45:42 (1 occurrences)
Last logged: 10:45:42

Setup failed for custom integration 'givenergy_local': Requirements for givenergy_local not found: ['pydantic>=1.10.12,<2'].

cdpuk, 2025-02-09:
Sounds like a bit of a dependency tangle, and your HA version simply isn't compatible with the integration version. I think if you'd updated the integration first, then HA, things might have been OK, but too late for that I guess!

It's unhelpful that there doesn't seem to be an easy way to fix this via the HACS UI. I've not had to deal with this myself, but perhaps something like this could help get it removed?

https://www.reddit.com/r/homeassistant/comments/s8ypd3/how_to_manually_remove_an_orphaned_repo_from_hacs/

garethedwards99, 2025-02-10:
Thanks for the update.
I deleted the repo using the custom_components folders using a samba share.  I have now successfully reinstalled the repo and the integration.

Thanks for your help

closed 2025-02-12 by cdpuk