cdpuk/givenergy-local #119: Slow Battery Charging

Swiftzn, 2025-03-24:
Hello,

Not sure if this is just a misconfiguration from my end or if this is something I have inadvertently caused my self.

Currently have a call with GivEnergy to see if they can find anything but currently my battery doesn't seem to be charging, it seems to hit the reserve %age (albeit very slowly) then not charge further.

Currently I am generating about 3.5kwh of excess but the battery is only charging at 290w (it is currently at 2% charge)

Below are my integration settings.

[image]

[image]

[image]

It's very possible this is something I have done just not sure what.
I also note the app has trouble reading from the inverter sometimes not sure if anyone has some suggestions

Swiftzn, 2025-03-24:
If I enable timed charge seems to work fine otherwise it gets no charge

cdpuk, 2025-06-21:
Just following up on some old issues. Everything looks correct in the configuration, so I'd expect it to be charging the battery at the maximum available rate. I'd be curious to hear the outcome of this - whether GivEnergy had anything to say or change?

Swiftzn, 2025-06-21:
> Just following up on some old issues. Everything looks correct in the configuration, so I'd expect it to be charging the battery at the maximum available rate. I'd be curious to hear the outcome of this - whether GivEnergy had anything to say or change?

Ooh meant to follow up on this, so had the installers come out and troubleshoot, and im not sure what they changed to be honest they only audited change i could see was a slight reduction in max charge rate which they dropped to like 2960w instead of 3000w wish I had a better answer for you